ros/meta-ros #730: configure error when for mrpt2 in meta-ros foxy

willofdiamond, 2020-08-30:
Hi I am trying to integrate Meta-ros foxy with my yocto build using CORE_IMAGE_EXTRA_INSTALL += "  ros-core  packagegroup-ros-world-foxy" in my local.conf for raspberry pi 3. By adding packagegroup-ros-world-foxy to CORE_IMAGE_EXTRA_INSTALL  I am receiving a configure error with mrpt2 . Not sure how to fix this issue

```
ERROR: mrpt2-2.0.4-1-r0 do_configure: Execution of '/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/temp/run.do_configure.6120' failed with exit code 1:
-- The C compiler identification is GNU 10.1.0
-- The CXX compiler identification is GNU 10.1.0
-- Check for working C compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-gcc
-- Check for working C compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++
-- Check for working CXX compiler: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-g++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/recipe-sysroot-native/usr/bin/pkg-config (found version "0.29.2")
-- Looking for timegm
-- Looking for timegm - found
-- Looking for _mkgmtime
-- Looking for _mkgmtime - not found
-- Looking for alloca
-- Looking for alloca - not found
-- Looking for gettid
-- Looking for gettid - found
-- Looking for sincos
-- Looking for sincos - found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for _aligned_malloc
-- Looking for _aligned_malloc - not found
-- Looking for alloca.h
-- Looking for alloca.h - found
-- Looking for linux/serial.h
-- Looking for linux/serial.h - found
-- Looking for linux/input.h
-- Looking for linux/input.h - found
-- Looking for malloc.h
-- Looking for malloc.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for winsock2.h
-- Looking for winsock2.h - not found
-- Looking for windows.h>
#include <winusb.h
-- Looking for windows.h>
#include <winusb.h - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long double
-- Check size of long double - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Searching 16 bit integer - Using unsigned short
-- Check if the system is big endian - little endian
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
[... 138 lines cut ...]
--  Phidgets                             :  No
--  RoboPeak LIDAR                       : Yes
--  SwissRanger 3/4000 3D camera         :  No
--  Videre SVS stereo camera             :  No
--  libudev (requisite for XSensMT)      : Yes (System)
--  xSENS MT                             : Yes (Built-in)
--  Intersense sensors                   :  No
--
-- Configuring incomplete, errors occurred!
See also "/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/build/CMakeFiles/CMakeOutput.log".
See also "/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/build/CMakeFiles/CMakeError.log".
WARNING: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/temp/run.do_configure.6120:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -Wno-dev'

ERROR: Logfile of failure stored in: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/temp/log.do_configure.6120
 WARNING: /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/temp/run.do_configure.6120:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/haradhyu/poky/noRosbuild/tmp/work/cortexa53-oe-linux/mrpt2/2.0.4-1-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -Wno-dev'
|
ERROR: Task (/home/haradhyu/poky/meta-ros/meta-ros2-foxy/generated-recipes/mrpt2/mrpt2_2.0.4-1.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 11037 tasks of which 10438 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/haradhyu/poky/meta-ros/meta-ros2-foxy/generated-recipes/libg2o/libg2o_2020.5.29-1.bb:do_configure
  /home/haradhyu/poky/meta-ros/meta-ros2-foxy/generated-recipes/mrpt2/mrpt2_2.0.4-1.bb:do_configure
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

shr-project, 2020-09-04:
Please always include at least "Build Configuration:" section shown at the beginning of the build and better describe the build configuration you've used.

mrpt2 isn't included in the default builds we're regularly testing (as defined by .mcf files and described in documentation). After adding commercial to LICENSE_FLAGS_WHITELIST I was able to reproduce this issue and will add a fix for it, but there are some other issues in mrpt2 as well, so you need mrpt2 (and all of packagegroup-ros-world-foxy) in your image?

shr-project, 2020-09-08:
Fixed in milestone 14.

closed 2020-09-08 by shr-project