ros/meta-ros #736: Can meta-ros run in petalinux?

nju044, 2020-09-09:
petalinux is a tool from Xilinx: https://www.xilinx.com/products/design-tools/embedded-software/petalinux-sdk.html

I currently plan to poring software to ZynqMP, but the software is developed in ubuntu+ROS+X86 env. So I want to first setup a similar system env(ubuntu+ROS+ARM). That's why I want to know if meta-ros can run in petalinux....

Any other suggestion is also welcomed. Thanks!

roeey777, 2020-10-03:
I haven't tried to work with `ros` on a `zyncmp` board, however I've already have worked with it on the `jetson tx2` board which also has arm based processor.
I don't see any reason why that shouldn't work since [petalinux](https://github.com/Xilinx/meta-petalinux) is just some layers from Xilinx.

nju044, 2020-10-04:
S> I haven't tried to work with `ros` on a `zyncmp` board, however I've already have worked with it on the `jetson tx2` board which also has arm based processor.
> I don't see any reason why that shouldn't work since [petalinux](https://github.com/Xilinx/meta-petalinux) is just some layers from Xilinx.

I came across pycryptodome and pycryptodomex conflict. Did you happen to know something about this? (Since the issue involves too much details, we can discuss it if you are interested or ever came across it....)

roeey777, 2020-10-04:
I didn't encountered such a conflict, which branches of `meta-ros` and `meta-petalinux` are you using?

shr-project, 2020-10-04:
For pycryptodome and pycryptodomex conflict see:
https://github.com/ros/meta-ros/commit/66f9be008070464de24198f3dde90cf88fda8124

if you don't really need to use old pycrypto, you should use just pycryptodomex and change whatever in Xilinx layers is pulling pycryptodome.

nju044, 2020-10-05:
> I didn't encountered such a conflict, which branches of `meta-ros` and `meta-petalinux` are you using?

meta-ros:  
      branch: origin/zeus 
      commit: 997f0fdcad0333dacdd4cdb185dd152ae2984492) 
PetaLinux:
      official release from xilinx, based on zeus version

nju044, 2020-10-05:
> For pycryptodome and pycryptodomex conflict see:
> [66f9be0](https://github.com/ros/meta-ros/commit/66f9be008070464de24198f3dde90cf88fda8124)
> 
> if you don't really need to use old pycrypto, you should use just pycryptodomex and change whatever in Xilinx layers is pulling pycryptodome.

Yes, I tried the fix you mentioned(change pycryptodome to pycryptodomex), then had following error:
`Error: Transaction check error:
  file /usr/lib/python2.7/site-packages/Crypto/Cipher/AES.py conflicts between attempted installs of python-pycrypto-2.6.1-r0.aarch64 and python-pycryptodome-3.4.11-r0.aarch64
  file /usr/lib/python2.7/site-packages/Crypto/Cipher/AES.pyc conflicts between attempted installs of python-pycrypto-2.6.1-r0.aarch64 and python-pycryptodome-3.4.11-r0.aarch64
  file /usr/lib/python2.7/site-packages/Crypto/Cipher/ARC2.py conflicts between attempted installs of python-pycrypto-2.6.1-r0.aarch64 and python-pycryptodome-3.4.11-r0.aarch64
  file /usr/lib/python2.7/site-packages/Crypto/Cipher/ARC2.pyc conflicts between attempted installs of python-pycrypto-2.6.1-r0.aarch64 and python-pycryptodome-3.4.11-r0.aarch64`

Really wierd. Any clues?

shr-project, 2020-10-05:
Check what depends on python-pycrypto and python-pycryptodome (without x) - use bitbake -g or you can temporary blacklist python-pycrypto and python-pycryptodome to see what will complain.

nju044, 2020-11-03:
> Check what depends on python-pycrypto and python-pycryptodome (without x) - use bitbake -g or you can temporary blacklist python-pycrypto and python-pycryptodome to see what will complain.

Sorry for the late reply, because the board just arrived last week. 
I found rosbag despends on python-pycryptodome. My workaround for this is to remove the dependency. So far so good(for building image). 

Is this ok?

shr-project, 2020-11-03:
> I found rosbag despends on python-pycryptodome.

It should depend on python-pycryptodomex since the meta-ros commit I've linked earlier. In long term, it would be better to eliminate pycrypto and pycryptodome dependencies, not pycryptodomex.

nju044, 2020-11-04:
> It should depend on python-pycryptodomex

My mistake. Thanks for pointing out.

closed 2020-11-04 by nju044