ros/meta-ros #737: meta-ros foxy (foonathan-memory) build failure on Yocto dunfill after update to milestone 14

rafaelhuang-intel, 2020-09-15:
Sharing build log:
build is for Yocto Dunfell.
both Ubuntu 18.04 & 20.04 all fail.
Metao-ros updated to milestone 14 (commit: a135ffca0a6717fbc30ce2fb2c07d1764c3218cc)
Issue is faced with foxy.

[foonathan-memory_U18.04_log.do_compile.27089.txt](https://github.com/ros/meta-ros/files/5222517/foonathan-memory_U18.04_log.do_compile.27089.txt)
[foonathan-memory_U20.04_log.do_compile.2719843.txt](https://github.com/ros/meta-ros/files/5222518/foonathan-memory_U20.04_log.do_compile.2719843.txt)

rafaelhuang-intel, 2020-09-15:
it's seens other buddy meet same issue,  https://github.com/ros/meta-ros/issues/727#issuecomment-689389436

shr-project, 2020-09-15:
What MACHINE are you building for? Does qemu work for you in other recipes?

rafaelhuang-intel, 2020-09-16:
my MACHIE is "intel-rt-corei7-64", and I tried add " EXTRA_OECMAKE += "-DFOONATHAN_MEMORY_BUILD_TESTS=OFF"
" in meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb and build again, another fail appear, log attached.

[foonathan-memory_U18.04_disable_build_tools_log.do_compile.520.txt](https://github.com/ros/meta-ros/files/5229169/foonathan-memory_U18.04_disable_build_tools_log.do_compile.520.txt)

shr-project, 2020-09-16:
https://layers.openembedded.org/layerindex/branch/master/machines/?q=intel-rt-corei7-64&search=1
doesn't show what "intel-rt-corei7-64" is exactly and you didn't answer the question about qemu, "Illegal instruction" is usually caused by incorrect configuration for qemu.

I don't know what other changes you did.
EXTRA_OECMAKE += "-DFOONATHAN_MEMORY_BUILD_TESTS=OFF"
is already there, I've asked you to disable TOOLS not TESTS (FOONATHAN_MEMORY_BUILD_TOOLS) which is the part which calls qemu.

And the examples build OK for all MACHINEs we support, you can try with qemux86-64 which should be relatively close to your "intel-rt-corei7-64" to see if it's caused by your MACHINE configuration or the DISTRO.

rafaelhuang-intel, 2020-09-17:
I changed "EXTRA_OECMAKE += "-DFOONATHAN_MEMORY_BUILD_TOOLS=OFF", it's my mistake, but build still fail. 
I will try "qemux86-64" to build .

shr-project, 2020-12-10:
Should be fixed with https://github.com/ros/meta-ros/pull/769 please test and reopen if it's still an issue.

closed 2020-12-10 by shr-project