ros/meta-ros #741: error caused by spdlog when integrating Meta-ros to existing yocto project

willofdiamond, 2020-09-16:
I am trying to integrate master branch of meta-ros with an existing yocto project. My main goal is to get  ros fox distro on to raspberry pi 3.  I had attached my BBlayer and local files for reference. 

**Error**
I am getting the following error.
```
ERROR: No recipes in default available for:
  /home/raj/poky/meta-ros/meta-ros-common/recipes-support/spdlog/spdlog_1.8.0.bbappend
```

**BBlayer**
```
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/raj/poky/meta \
  /home/raj/poky/meta-poky \
  /home/raj/poky/meta-yocto-bsp \
  /home/raj/poky/meta-openembedded/meta-oe \
  /home/raj/poky/meta-openembedded/meta-xfce \
  /home/raj/poky/meta-openembedded/meta-gnome \
  /home/raj/poky/meta-openembedded/meta-multimedia \
  /home/raj/poky/meta-openembedded/meta-networking \
  /home/raj/poky/meta-openembedded/meta-python \
  /home/raj/poky/meta-raspberrypi \
  /home/raj/poky/meta-openembedded/meta-filesystems \
  /home/raj/poky/meta-ros/meta-ros-common \
  /home/raj/poky/meta-ros/meta-ros2 \
  /home/raj/poky/meta-ros/meta-ros2-foxy \
  "

#DISTRO= 'ros2'
#ROS_DISTRO = 'foxy'
#OE_RELEASE_SERIES ='dunfell'

BBFILE_PRIORITY_raspberrypi_forcevariable = "50"
BBFILE_PRIORITY_ros2-foxy-layer_forcevariable = "37"
BBFILE_PRIORITY_ros1-layer_forcevariable = "36"
BBFILE_PRIORITY_ros-common-layer_forcevariable = "35"
BBFILE_PRIORITY_meta-python_forcevariable = "13"
BBFILE_PRIORITY_openembedded-layer_forcevariable = "10"
BBFILE_PRIORITY_core_forcevariable = "5"

# Lines from the BblayersConfExtraLines setting:
MCF_DISTRO ?= "ros2"
ROS_DISTRO ?= "foxy"
MCF_SUPPORTED_MACHINES ?= "qemux86 raspberrypi3"
MCF_OPENEMBEDDED_VERSION ?= "3.1~master-dunfell"
MCF_OE_RELEASE_SERIES ?= "dunfell"
ROS_PYTHON_VERSION ?= "3"
ROS_OE_RELEASE_SERIES ?= "dunfell"
```

**Local file**
```
#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
MACHINE ??= "raspberrypi3-64"

#Make faster
BB_NUMBER_THREADS ='32'
PARALLEL_MAKE = '-j 32'

EXTRA_IMAGE_FEATURES = "debug-tweaks package-management"
#CORE_IMAGE_EXTRA_INSTALL += "gnupg htop libgcc dpkg ros-core  boost packagegroup-ros-world-foxy"
CORE_IMAGE_EXTRA_INSTALL += "gnupg htop libgcc dpkg boost"

IMAGE_INSTALL_append = "gnupg wpa-supplicant git apt sudo"

#Adding glfw and opengl
DISTRO_FEATURES_append = " dpkg gnupg glfw opengl"

LICENSE_FLAGS_WHITELIST = "commercial"
SSTATE_MIRRORS = "\
     file://.* http://sstate.yoctoproject.org/dev/PATH;downloadfilename=PATH \n \
     file://.* http://sstate.yoctoproject.org/3.0.2/PATH;downloadfilename=PATH \n \
     file://.* http://sstate.yoctoproject.org/3.1.1/PATH;downloadfilename=PATH \n \
     "

DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_ipk"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

USER_CLASSES ?= "buildstats image-mklibs image-prelink"

PATCHRESOLVE = "noop"

#
# Disk Space Monitoring during the build
#
# Monitor the disk space during the build. If there is less that 1GB of space or less
# than 100K inodes in any key build location (TMPDIR, DL_DIR, SSTATE_DIR), gracefully
# shutdown the build. If there is less that 100MB or 1K inodes, perform a hard abort
# of the build. The reason for this is that running completely out of space can corrupt
# files and damages the build in ways which may not be easily recoverable.
# It's necesary to monitor /tmp, if there is no space left the build will fail
# with very exotic errors.
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"

# ROS-ADDITIONS-BEGIN
# ^^^^^^^^^^^^^^^^^^^ In the future, tools will expect to find this line.

# Increment the minor version whenever you add or change a setting in this file.
ROS_LOCAL_CONF_SETTINGS_VERSION := "2.2"

# If not using mcf, replace ${MCF_DISTRO} with the DISTRO being used.
DISTRO ?= "${MCF_DISTRO}"

# If not using mcf, set ROS_DISTRO in conf/bblayers.conf .

# The list of supported values of MACHINE is found in the Machines[] array in the .mcf file for the selected configuration.
# Use ?= so that a value set in the environment will override the one set here.
#MACHINE ?= "<SUPPORTED-MACHINE>"

# Can remove if DISTRO is "webos". If not using mcf, replace ${MCF_OPENEMBEDDED_VERSION} with the version of OpenEmbedded
# being used. See the comments in files/ros*.mcf for its format.
ROS_DISTRO_VERSION_APPEND ?= "+${MCF_OPENEMBEDDED_VERSION}"

# Can remove if DISTRO is not "webos". If not using mcf, replace ${MCF_WEBOS_BUILD_NUMBER} with the build number of webOS OSE
# being used.
#ROS_WEBOS_DISTRO_VERSION_APPEND ?= ".${MCF_WEBOS_BUILD_NUMBER}"

ROS_OE_RELEASE_SERIES_SUFFIX ?= "-${ROS_OE_RELEASE_SERIES}"
# Because of a bug in OpenEmbedded, <ABSOLUTE-PATH-TO-DIRECTORY-ON-SEPARATE-DISK> can not be a symlink.
#ROS_COMMON_ARTIFACTS ?= "<ABSOLUTE-PATH-TO-DIRECTORY-ON-SEPARATE-DISK>"

# Set the directories where downloads, shared-state, and the output from the build are placed to be on the separate disk.
#DL_DIR ?= "${ROS_COMMON_ARTIFACTS}/downloads"
#SSTATE_DIR ?= "${ROS_COMMON_ARTIFACTS}/sstate-cache${ROS_OE_RELEASE_SERIES_SUFFIX}"
#TMPDIR ?= "${ROS_COMMON_ARTIFACTS}/BUILD-${DISTRO}-${ROS_DISTRO}${ROS_OE_RELEASE_SERIES_SUFFIX}"
# Don't add the libc variant suffix to TMPDIR.
#TCLIBCAPPEND := ""

# As recommended by https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#var-BB_NUMBER_THREADS
# and https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#var-PARALLEL_MAKE:
#BB_NUMBER_THREADS ?= "${@min(int(bb.utils.cpu_count()), 20)}"
#PARALLEL_MAKE ?= "-j ${BB_NUMBER_THREADS}"

# Reduce the size of the build artifacts by removing the working files under TMPDIR/work. Comment this out to preserve them
# (see https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#ref-classes-rm-work).
INHERIT += "rm_work"

# Any other additions to the file go here.

# EXTRA_IMAGE_FEATURES is just one of the many settings that can be placed in this file. You can find them all by searching
# https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html#ref-variables-glossary for "local.conf".

# Uncomment to allow "root" to ssh into the device. Not needed for images with webOS OSE because it implicitly adds this
# feature.
# EXTRA_IMAGE_FEATURES += "ssh-server-dropbear"

# Uncomment to include the package management utilities in the image ("opkg", by default). Not needed for images with
# webOS OSE because it implicitly adds this feature.
# EXTRA_IMAGE_FEATURES += "package-management"

# Uncomment to have all interactive shells implicitly source "setup.sh" (ROS 1) or "ros_setup.sh" (ROS 2).
# EXTRA_IMAGE_FEATURES += "ros-implicit-workspace"

# Uncomment to display additional useful variables in the build configuration output.
# require conf/distro/include/ros-useful-buildcfg-vars.inc

# vvvvvvvvvvvvvvvvv In the future, tools will expect to find this line.
# ROS-ADDITIONS-END

```

shr-project, 2020-09-16:
```
ERROR: No recipes in default available for:
  /home/raj/poky/meta-ros/meta-ros-common/recipes-support/spdlog/spdlog_1.8.0.bbappend
```

what meta-ros revision are you building? current dunfell and master branch as spdlog_1.7.0.bbappend.

spdlog_1.8.0.bbappend is only in my shr-project fork and in this PR:
https://github.com/ros/meta-ros/pull/740
if you're trying to build from this PR, then you need to have corresponding newer meta-oe from
https://github.com/ros/meta-ros/pull/739
and if you're really building from dunfell branches, then to enable meta-ros-backports-gatesgarth layer in your bblayers.conf.

See
https://github.com/ros/meta-ros/commit/9ee61883026bcd2a88a21683591177fa4674e523
https://github.com/ros/meta-ros/pull/739/commits/e6da3c108c17c8b15b5651bcea80583fa08ebb80
https://github.com/ros/meta-ros/commit/0d2a86e1937cc71a3eb04568be6936085a782101

willofdiamond, 2020-09-16:
My current revision is dunfell/milestones/14-21-g433e7808c

shr-project, 2020-09-16:
433e7808c existed shortly only in my fork on shr-project.

You need to use it with the new meta-ros-backports-gatesgarth layer as mentioned before.

And it would be better to switch to some revision which exists in one of ros/meta-ros branches (433e7808c doesn't)
```
$ git branch -a --contains 433e7808c
$
```

willofdiamond, 2020-09-16:
Enabling meta-ros-backports-gatesgarth layer in bblayer worked

closed 2020-09-16 by willofdiamond