ros/meta-ros #742: rcl-logging-spdlog compile  failure with ros2 foxy

willofdiamond, 2020-09-17:
My main goal is to get ros fox distro on to raspberry pi 3. Local and bbfiles  can be found in #741. I am using 
`CORE_IMAGE_EXTRA_INSTALL += " ros-core   packagegroup-ros-world-foxy" `to get ros-core and recipes in packagegroup-ros-world-foxy to integrate into my image.  I had recived an compilation error with rcl-logging-spdlog compile  failure. I had attached the log file for the error.

**Error**
Received a do compile error rcl-logging-spdlog while trying to install 

`ERROR: Task (/home/haradhyu/poky/meta-ros/meta-ros2-foxy/generated-recipes/rcl-logging/rcl-logging-spdlog_1.0.1-1.bb:do_compile) failed with exit code '1'`

[log.do_compile.txt](https://github.com/ros/meta-ros/files/5236128/log.do_compile.txt)

shr-project, 2021-04-12:
@willofdiamond can you re-try with current version of layers? rcl-logging-spdlog still builds fine for me including aarch64 builds I don't see any obvious version why it would fail for you like this. You can also try to replace poky DISTRO with the nodistro config and if it fails only with poky, then you can compare what's different in those 2 builds.

shr-project, 2021-04-12:
well this setup is a bit strange, meta-oe revision 698748c1538ed03efbcfdd936cf8317b4f138c29 is from gatesgarth branch (which would suggest that meta-ros-backports-gatesgarth shouldn't be included), while your meta-ros revision a135ffca0a6717fbc30ce2fb2c07d1764c3218cc is from dunfell.

Make sure that you're fetching all layers from the same branch (like our .mcf files in build branch do).

closed 2021-04-12 by shr-project