ros/meta-ros #743: Navigation2 foxy launch fail on Yocto dunfill

rafaelhuang-intel, 2020-09-17:
Navigation2 foxy launch fail on Yocto dunfill

Navigation2 version:
nav2-bringup                                      :0.4.1-1-r0
navigation2                                       :0.4.1-1-r0

Run command:
ros2 launch nav2_bringup bringup_launch.py use_sim_time:=True autostart:=True map:=/home/root/test/map/map.yaml 

fail log:
[navigation2_simulate_error_log_20200917_1.txt](https://github.com/ros/meta-ros/files/5237584/navigation2_simulate_error_log_20200917_1.txt)

after check upstream, and found same issue ( https://github.com/ros-planning/navigation2/issues/1865 ) and PR ( https://github.com/ros-planning/navigation2/pull/1868/files ).

I tried manually change files (refer PR 1868) and run again, new issue appear,

fail log:

[navigation2_simulate_error_log_20200917_2.txt](https://github.com/ros/meta-ros/files/5237620/navigation2_simulate_error_log_20200917_2.txt)

rafaelhuang-intel, 2020-09-23:
I see PR734 ( https://github.com/ros/meta-ros/pull/734 ) and what time will merge to Dunfill branch?

rafaelhuang-intel, 2020-09-23:
I tried manually change local navigation2, nav2_bringup relate package to 0.4.3-1, build are passed, but if launch navigation2, still fail, log attached at below.

[navigation2_simulate_error_log_20200923_1.txt](https://github.com/ros/meta-ros/files/5266131/navigation2_simulate_error_log_20200923_1.txt)

rafaelhuang-intel, 2020-10-14:
https://github.com/ros2-gbp/rosidl_typesupport_fastrtps-release/issues/4

rafaelhuang-intel, 2020-10-21:
@shr-project 
I local merge PR734 ( #734 ) and I tried build on Yocto, but meet an error that "nav2_bringup" add in meta-ros2-foxy blacklist, so does it's means Yocto/Dunfill meta-ros2-foxy not support yet?

and then, I commment this blacklist and build again "nav2_bringup", all build pass but fail at run nav2_bringup, below is fail log.

[navigation2_simulate_error_log_20201021_1.txt](https://github.com/ros/meta-ros/files/5412705/navigation2_simulate_error_log_20201021_1.txt)

shr-project, 2020-10-21:
PR734 is a draft PR as work in progress and there are still quite a few issues to resolve before it's ready for merge. I don't know when I'll be able to resolve them myself, if you care about this PR, please send fixes for remaining issues (listed in the PR) to make it ready sooner.

meta-ros2-foxy as is in dunfell branch is the support for Yocto/Dunfell, but that doesn't imply that all the ROS components were tested in runtime, you can see the performed testing in https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#ros-2-sanity-test and the "Done when" section in https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-14

nav2_bringup blacklist in PR734 is defined as:
```
PNBLACKLIST[nav2-bringup] ?= "${@bb.utils.contains_any('ROS_WORLD_SKIP_GROUPS', ['qt5', 'x11'], 'qt5: depends on slam-toolbox which depends on qtbase; x11: depends on slam-toolbox which depends on rviz-common->rviz-ogre-vendor->(libx11,libxrandr,libxaw) which require x11 in DISTRO_FEATURES', '', d)}"
```
if it's blacklisted for you then either you don't have meta-qt5 in your layers or x11 in DISTRO_FEATURES, if you just remove the blacklist, then it's not surprising that it fails to build for you.

shr-project, 2021-04-12:
https://github.com/ros/meta-ros/pull/734 was merged long time ago, not sure what's left for this ticket, closing. Open new one if something else is needed.

closed 2021-04-12 by shr-project