ros/meta-ros #748: ROS2 eloquent dunfell: ros-image-core does not build for MACHINE = 'intel-corei7-64'

elektrokokke, 2020-10-14:
Hi!

When building the ros-image-core image for ROS2 eloquent (on dunfell), everything works fine.
But when trying to build for the intel-corei7-64 MACHINE from the meta-intel layer, foonathan-memory fails with the attached log message.

[log.do_compile.11164.log](https://github.com/ros/meta-ros/files/5374794/log.do_compile.11164.log)

qemu crashed due to an illegal instruction.

I do not know, what foonathan-memory is for exactly (I suspect something with fastrtps). And I don't know why qemu is involved at all. It is also unclear to me why the only two supported machines (in the meta-ros .mcs files) are qemux86-64 and raspberrypi4-64. What do I have to do to build for a certain processor?

Thanks a lot for any help on this!

Cheers,
 elektrokokke

shr-project, 2020-10-16:
Looks the same as #737

shr-project, 2020-10-16:
Can you try to add ${QEMU_OPTIONS} to

EXTRA_OECMAKE = '-DCMAKE_CROSSCOMPILING_EMULATOR="${@qemu_target_binary(d)} ${QEMU_OPTIONS}"'
in meta-ros2/recipes-devtools/foonathan-memory/foonathan-memory_git.bb?

That should pass "-r ${OLDEST_KERNEL} -cpu Nehalem,check=false" for your intel-corei7-64 MACHINE when foonathan calls qemu.

For more details about this you can read:
https://www.openembedded.org/pipermail/openembedded-core/2018-April/150178.html
https://www.openembedded.org/pipermail/openembedded-core/2018-April/268612.html

> It is also unclear to me why the only two supported machines (in the meta-ros .mcs files) are qemux86-64 and raspberrypi4-64.

Because we don't have resources to build and test meta-ros with every single MACHINE supported by OE. We have select 4 well supported MACHINEs from 4 different architectures as documented in:
https://github.com/ros/meta-ros/wiki/Policies#machine-support

But it's easy to add extra BSP layers and build for other MACHINEs, it's just not built or tested when we do meta-ros changes.

elektrokokke, 2020-10-25:
Thanks a lot, this worked with a slight change: I had to use

EXTRA_OECMAKE = '-DCMAKE_CROSSCOMPILING_EMULATOR="${@qemu_target_binary(d)};-r;3.2.0;-cpu;Nehalem,check=false"'

CMAKE_CROSSCOMPILING_EMULATOR expects arguments as a semicolon-separated list. Strings with spaces will be enclosed in quotes, which makes the command fail.

There is probably a way to get the right string using ${QEMU_OPTIONS}, but this worked as a quick fix for me.

closed 2020-12-10 by shr-project