ros/meta-ros #749: Moveit2 foxy launch fail on Yocto dunfill

rafaelhuang-intel, 2020-10-14:
Moveit2 foxy launch fail on Yocto dunfill 

Moveit2 version:
Moveit : 1.1.0-r0
run-moveit-cpp :2.0.0-r0

Run Comamnd:

ros2 launch run_moveit_cpp run_moveit_cpp.launch.py

or
ros2 launch run_moveit_cpp run_moveit_cpp_fake_controller.launch.py

Fail log:

[foxy_moveit_launch_fail_on_yocto_20201013.txt](https://github.com/ros/meta-ros/files/5375468/foxy_moveit_launch_fail_on_yocto_20201013.txt)

rafaelhuang-intel, 2020-10-14:
relate issue found on Navigation2 : https://github.com/ros/meta-ros/issues/743

rafaelhuang-intel, 2020-10-14:
https://github.com/ros2-gbp/rosidl_typesupport_fastrtps-release/issues/4

shr-project, 2020-10-22:
For "undefined symbols" issues, I'm aware of this, it was reported on github for other components as well, 

and I was working on similar issue in the initial foxy release, which resulted to this:https://github.com/ros/meta-ros/commit/d80f1ed75a55a059ba28524ff9657e306e0912f1
and couple following commits to resolve this error in "ros2 topic list"

```
root@qemux86-64:~# ros2 topic list

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'type support not from this implementation, at /usr/src/debug/rmw-fastrtps-cpp/1.0.0-1-r0/git/src/publisher.cpp:75, at /usr/src/debug/rcl/1.1.2-1-r0/git/src/rcl/node.c:276'

with this new error message:

  'rcl node's rmw handle is invalid, at /usr/src/debug/rcl/1.1.2-1-r0/git/src/rcl/node.c:428'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1591122660.978804206] [rcl]: Failed to fini publisher for node: 1
Unknown error creating node: rcl node's rmw handle is invalid, at /usr/src/debug/rcl/1.1.2-1-r0/git/src/rcl/node.c:428
```

And while working on this I remember seeing many cases of undefined symbols whenever some component accidentally linked against native dependency in RSS instead the target one, that might be the case for your build as well.
Unfortunately we were planning to debug and resolve this more properly (by changing how the ros workspace is created) together with the work on ROS2 distros using /opt prefix like ROS1 does, which is currently listed for Milestone 15, but it will be pushed to Milestone 16 or later, because of other tasks:
https://github.com/ros/meta-ros/wiki/Superflore-OE-Recipe-Generation-Scheme#milestone-15

Sorry, I cannot guess when I'll get time to go back to this. It would be great if you can debug and fix it yourself (patches are very welcome), but don't expect it to be easy (unless the root cause is different from what I was seeing before).

LewisLiuPub, 2020-11-02:
@shr-project , We have fixed some errors when building/launching moveit2 packages, it's my pleasure to upload our contributions to this project. There are some preconditions I need your help to provide some solutions or hints:
* code repo and release branches of Moveit2. Have you or someone else already created them? which one can I use?
* which branch of meta-ros could I use for my PR?

Thanks.

shr-project, 2020-11-02:
> code repo and release branches of Moveit2. Have you or someone else already created them? which one can I use?

If the issue is in moveit2 then upload it to upstream moveit2 repos

Then add it as .patch file in meta-ros with
`Upstream-Status: Submitted [link to the PR in upstream repo]`
and meta-ros PR can target dunfell branch if that's what you're using, I'll merge it to gatesgarth and master branches after it's tested and merged with dunfell builds.