ros/meta-ros #763: spdlog bbappend has wrong version in dunfell branch

windelbouwman, 2020-11-17:
**Describe the bug**

When building ros2 against the openembedded `dunfell` branch, `spdlog_1.8.1.bbappend` does not match any recipe. I renamed this to `spdlog_1.5.0.bbappend` to continue.

**To Reproduce**

Layers used:
- openembedded-core/meta (dunfell branch)
- meta-openembedded/meta-python (dunfell branch)
- meta-openembedded/meta-oe (dunfell branch)
- meta-ros/meta-ros-common (dunfell branch)
- meta-ros/meta-ros2 (dunfell branch)
- meta-ros/meta-ros2-dashing (dunfell branch)

Command used:

    $ bitbake ros2topic

shr-project, 2020-11-17:
That's because you're missing meta-ros-backports-gatesgarth layer, see
https://github.com/ros/meta-ros/commit/6afd5676b53de371ecc57ae641b17f297d12f4c1
and
https://github.com/ros/meta-ros/commit/880ae22173ca037ecd76fbf103cc3aa908f46f48

windelbouwman, 2020-11-17:
Ah, okay, I see! Thanks for the tip!

I think it would make sense to add a small howto to the README of this repo how to use it. For example:

# Usage

Clone this repository, using the proper branch for your yocto version

    $ git clone https://github.com/ros/meta-ros -b dunfell

Add these layers to your `bblayers.conf` file:

```
meta-ros/meta-ros-backports-gatesgarth
meta-ros/meta-ros-common
meta-ros/meta-ros2
meta-ros/meta-ros2-dashing
```

Create an image by your preferred way, for example set `IMAGE_INSTALL_append` in you `local.conf` file:

```
IMAGE_INSTALL_append = " rclpy"
```

Would it make sense to add this? I was unable to find this usage information so far.

windelbouwman, 2020-11-17:
Btw, what does `gatesgarth` signify?

shr-project, 2020-11-17:
Yocto 3.1 release codename is Gatesgarth as mentioned in commit message of the second commit.

For all releases see
https://wiki.yoctoproject.org/wiki/Releases

There is a README file which points to
https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions
which describes how to use mcf to get the right set of layers in right revisions (to match exactly what we have tested), if you use different set of layers or different versions then you're bit on your own, but LAYERDEPENDS should guide you about which layers you need, unfortunately this meta-ros-backprots-gatesgarth layer is a bit of exception, because until now we were using identical branch content for dunfell, gatesgarth, hardknott, that will probably change soon and then I'll add meta-ros-backports-gatesgarth to LAYERDEPENDS in dunfell branch as well (and remove the layer completely from gatesgarth and master branches).

windelbouwman, 2020-11-23:
> I'll add meta-ros-backports-gatesgarth to LAYERDEPENDS in dunfell branch as well (and remove the layer completely from gatesgarth and master branches)

This makes a lot of sense! Thank you for shaping these layers in its current shape!

windelbouwman, 2020-11-23:
I am able to build the recipe, but when running ros2 topic list, I get: "Unable to get rmw_implementation_identifier, try specifying the implementation to use via the 'RMW_IMPLEMENTATION' environment variable", and hints on this?

windelbouwman, 2020-11-23:
Interesting --> https://github.com/ros2/rcpputils/issues/40

This might be worth a note.

closed 2020-11-24 by shr-project

herb-kuta-lge, 2020-11-25:
> I think it would make sense to add a small howto to the README of this repo how to use it. 

Yeah, perhaps the

> See [here](https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions)
for instructions.

is a little too terse.

windelbouwman, 2020-11-25:
Also, I was unable to locate the absolute coolest use-case (in my opinion) of the repo:

1. Inlcude layers in bblayers
2. add package to image
3. profit :)

Benefits of storing the usage in the readme, is that for each branch, there are subtle differences in backports / usage etc.. The wiki might get easy out of sync. The wiki is very useful on it's own, but I'll try to cook up some readme improvements!