ros/meta-ros #766: Make it easier to remove packages depending on fortran enabled in gcc

shr-project, 2020-11-24:
As discussed in:
https://github.com/ros/meta-ros/commit/a2abbc6022c89cadf2900522ee0b0616acf6d6ba#commitcomment-44438887

People are struggling how to build fortran on their platform (it builds OK for all MACHINEs we're testing):
https://bugzilla.yoctoproject.org/show_bug.cgi?id=14113

Add support for "fortran" in ROS_WORLD_SKIP_GROUPS which will control including conf/ros-distro/include/enable-fortran.inc from ros-distro.conf as well as blacklisting recipes like lapack, fetch-ikfast-plugin.

It won't be included by default.

shr-project, 2020-11-24:
@nju044 FYI

nju044, 2020-12-05:
> @nju044 FYI

Thank you very much.
The hash id of my meta-ros is  as follows:
```
5707a46 - (HEAD -> zeus, origin/zeus-next, origin/zeus) packagegroup-ros2-demos: inherit ros_distro_${ROS_DISTRO} (6 weeks ago) 
        - Martin 'JaMa' Jansa <Martin.Jansa@gmail.com>     2020-10-11 14:35:02 -0700| 
```

This means your comment hasn't been merged into master or zeus. But I wanna try it.
My plan is to get latest zeus branch, then merge yours into it, then merge my modification into it. Is it OK?

nju044, 2020-12-05:
Hi Jansa
I wanna merge jansa-766 to my local meta-ros. Then a few questions rise, please help me out:
1. I chekout your branch jansa-766. There are so many changes ahead of zeus. I don't think I can manage the merge.
2. I notice jansa-766 includes a new layer called meta-ros-python2(also in master branch), which doesn't exist in branch zeus. Do I need to put this layer in zeus? What is it for?
3. continued with 1 ==> after comparing master with jansa-766, three files changed:
     meta-ros1-melodic/conf/ros-distro/include/melodic/ros-distro-recipe-blacklist.inc
     meta-ros1-melodic/recipes-core/packagegroups/packagegroup-ros-world-melodic.bb
     meta-ros-common/conf/ros-distro/include/enable-fortran.inc
Merging these three files from jansa-766 to zeus is more complicated than that to master. So what's the difference between master and zeus? Do you suggest using mater or zeus?

shr-project, 2020-12-05:
{quote}
Is that adequate? What's the difference between master and zeus? Do you suggest using mater or zeus?
{quote}

master branch is for Yocto 3.3 Hardknott, zeus is for Yocto 3.0 Zeus, see:
https://wiki.yoctoproject.org/wiki/Releases

I wouldn't recommend master for production (as it's development branch of Yocto) and I wouldn't recommend Zeus as well as it's out of official support already.

Current best release for production is LTS Yocto 3.1 Dunfell.

{quote}
I need to put this layer in zeus? What is it for?
{quote}

See the commit which introduced it:
https://github.com/ros/meta-ros/commit/7ec08231b530ac5f3be55171fabe234e7b8ebe2e

You probably don't need it in zeus as we don't plan to support Noetic in zeus.

nju044, 2020-12-07:
> I wouldn't recommend master for production (as it's development branch of Yocto) and I wouldn't recommend Zeus as well as it's out of official support already.
> 
> Current best release for production is LTS Yocto 3.1 Dunfell.
> 
Well, I guess I have to stick with Zeus because it's part of offically released BSP from the board vendor.

I merged your commit into my local meta-ros, but the previous libfortran issue still there. And fcl/pcl failure occured.
The whole system seems like a puzzle to me. Do you happen to know someone who have integrated meta-ros and zeus before?

shr-project, 2020-12-07:
> Do you happen to know someone who have integrated meta-ros and zeus before?

It still builds fine here (for what we're testing), the issue is specific to your BSP.

I still haven't seen answer about world from:
https://github.com/ros/meta-ros/commit/a2abbc6022c89cadf2900522ee0b0616acf6d6ba#commitcomment-44378229
if you don't need whole world (you probably don't) then you can just skip more packages which fail with xilinx's modifications (it might be impossible to skip pcl, you'll have to debug this one (and pcl is sometimes picky to build right).

nju044, 2020-12-07:
> It still builds fine here (for what we're testing), the issue is specific to your BSP.
Can't agree more.
I initiate a call for meta-ros/yocto/petalinux opensource project in xilinx forum, but there's no feedback :(

> if you don't need whole world (you probably don't) then you can just skip more packages which fail with xilinx's modifications (it might be impossible to skip pcl, you'll have to debug this one (and pcl is sometimes picky to build right).
I've already done that. The fortran issue occured after I remove some recipes. 
Right now, I'm a newbie to yocto, all I do is to use petalinux to build image. Beside, the bsp is also a blind spot for me, as I cann't debug it. All these make things complicated. Maybe I should first spend some time on yocto, then meta-ros without bsp, finally with bsp.

shr-project, 2020-12-07:
I've replied in yocto bugzilla, the log you shared there reveals what caused this issue in pentalinux build, read:
https://bugzilla.yoctoproject.org/show_bug.cgi?id=14113#c22

With mcf from https://github.com/shr-project/meta-ros/commit/1265eb7b0dd5f13ff29f603527097b954dfdeab9
where I've replaced meta-raspberrypi with meta-xilinx-bsp, I was able to build libgfortran and lapack for zcu102-zynqmp MACHINE just fine

```
ros1-melodic-zeus-xilinx$ export MACHINE=zcu102-zynqmp
ros1-melodic-zeus-xilinx$ bitbake libgfortran
Parsing recipes: 100% |#################################################################################################################################################################################################################################| Time: 0:00:25
Parsing of 4012 .bb files complete (0 cached, 4012 parsed). 5172 targets, 811 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "zcu102-zynqmp"
DISTRO               = "ros1"
DISTRO_VERSION       = "1.melodic.20200322.7-13+3.0-411-zeus"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System (ROS 1) with OpenEmbedded Linux"
ROS_DISTRO           = "melodic"
ROS_VERSION          = "1"
ROS_PYTHON_VERSION   = "2"
meta-xilinx-bsp      = "zeus:b82343ac5f013926839627cee9dae7106c008ae9"
meta-ros1-melodic
meta-ros1
meta-ros-common
meta-ros-backports-dunfell = "zeus-next:5b1b580d592e78cb4ee0996844d017b15dd76459"
meta-python
meta-oe              = "zeus:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta                 = "zeus:db8ceed8f2eca92a4cffe8295481d8041281fdd0"

Initialising tasks: 100% |##############################################################################################################################################################################################################################| Time: 0:00:01
Checking sstate mirror object availability: 100% |######################################################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 38 Found 5 Missed 33 Current 102 (13% match, 76% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Tasks Summary: Attempted 613 tasks of which 527 didn't need to be rerun and all succeeded.
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 25.0% sstate reuse(2 setscene, 6 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 5 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 5 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 5 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 5 scratch)
NOTE:   do_populate_lic: 25.0% sstate reuse(2 setscene, 6 scratch)
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds
```

```
ros1-melodic-zeus-xilinx$ bitbake lapack
Loading cache: 100% |###################################################################################################################################################################################################################################| Time: 0:00:01
Loaded 5171 entries from dependency cache.
Parsing recipes: 100% |#################################################################################################################################################################################################################################| Time: 0:00:01
Parsing of 4012 .bb files complete (4011 cached, 1 parsed). 5172 targets, 811 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "zcu102-zynqmp"
DISTRO               = "ros1"
DISTRO_VERSION       = "1.melodic.20200322.7-13+3.0-411-zeus"
TUNE_FEATURES        = "aarch64"
[... 5 lines cut ...]
meta-xilinx-bsp      = "zeus:b82343ac5f013926839627cee9dae7106c008ae9"
meta-ros1-melodic
meta-ros1
meta-ros-common
meta-ros-backports-dunfell = "zeus-next:5b1b580d592e78cb4ee0996844d017b15dd76459"
meta-python
meta-oe              = "zeus:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta                 = "zeus:db8ceed8f2eca92a4cffe8295481d8041281fdd0"

Initialising tasks: 100% |##############################################################################################################################################################################################################################| Time: 0:00:00
Checking sstate mirror object availability: 100% |######################################################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 16 Found 10 Missed 6 Current 134 (62% match, 96% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
NOTE: Tasks Summary: Attempted 648 tasks of which 631 didn't need to be rerun and all succeeded.
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_package_qa: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE:   do_package_write_ipk: 83.3% sstate reuse(5 setscene, 1 scratch)
NOTE:   do_populate_lic: 0.0% sstate reuse(0 setscene, 1 scratch)
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 1 seconds
```

closed 2020-12-10 by shr-project