ros/meta-ros #788: turtlesim_1.2.5 not building on meta-ros2

munoz0raul, 2021-01-08:
Hi all, I'm trying to build turtlesim + qt but I had an issue that I couldn't find a solution.

```
CMake Error at CMakeLists.txt:45 (qt5_wrap_cpp):
  Unknown CMake command "qt5_wrap_cpp".
```

I'm using meta-ros2-foxy on gatesgarth. It works fine without turtlesim.

Any idea on how to fix it?

shr-project, 2021-01-20:
See corresponding fix in dashing:
https://github.com/ros/meta-ros/commit/3faebc2a24d5b3ac5e57d960cbc3ea2a2d652c37

munoz0raul, 2021-01-20:
Hello @shr-project, I manage fix the issue with this bbappend:

`inherit cmake qmake5_paths
inherit populate_sdk_qt5

EXTRA_OECMAKE = "-DOE_QMAKE_PATH_EXTERNAL_HOST_BINS='${OE_QMAKE_PATH_EXTERNAL_HOST_BINS}'"

ROS_BUILDTOOL_EXPORT_DEPENDS_prepend = " \
    ament-cmake-native \
    rosidl-cmake-native \
"

DEPENDS += "qtbase qtbase-native rosidl-typesupport-fastrtps-cpp-native rosidl-typesupport-fastrtps-c-native"
`

After qt errors, I also got some missing dependencies relates to fastrtps-c and cpp.

closed 2021-07-16 by shr-project