ros/meta-ros #793: INTERFACE_LINK_LIBRARIES in SDK does include yocto build path

npnmos, 2021-02-04:
Hello, 
I am using the Zeus branch that i understand is no more supported.
I have generated an SDK for my image. I then tried to compile a "hello world" for fastRTPS and got errors because it was not finding tinyxml2 in a certain path. The strange thing was that it was pointing to the build folder of fastrtps inside yocto and not to the SDK folder. Looking into `<sdk-folder>/sysroots/aarch64-poky-linux/usr/share/fastrtps/cmake/fastrtps-targets.cmake` I saw this:

```
set_target_properties(fastrtps PROPERTIES
   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
   INTERFACE_LINK_LIBRARIES "fastcdr;-lpthread;dl;/home/yocto/imx-yocto-L5.4.24_2.1.0/build-inm034r10/tmp/work/aarch64-poky-linux/fastrtps/1.8.2-1-r0/recipe-sysroot/usr/lib/libtinyxml2.so;\$<\$<BOOL:1>:   OpenSSL::SSL\$<SEMICOLON>OpenSSL::Crypto>;\$<\$<BOOL:>:iphlpapi\$<SEMICOLON>Shlwapi>"
 )
```
and it is pointing to my build directory. I fixed by writing this and pointing to the SDK:

```
set_target_properties(fastrtps PROPERTIES
   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include"
   INTERFACE_LINK_LIBRARIES "fastcdr;-lpthread;dl;/home/yocto/sdk/fsl-imx-xwayland/sysroots/aarch64-poky-linux/usr/lib/libtinyxml2.so;\$<\$<BOOL:1>:OpenSSL::SSL\$<SEMICOLON>OpenSSL::Crypto>;\$<\$<BOOL:>:  iphlpapi\$<SEMICOLON>Shlwapi>"
 )
```
Sorry if it was not very clear but I do not totally understand Cmake.

hannes09, 2021-02-25:
I have found a similar problem like this also with `rcl-yaml-param-parser`,  in an sdk build librcl ->librcl_yaml_param_parser  has a hardcoded wrong path in `rcl_yaml_param_parser/cmake/rcl_yaml_param_parserExport.cmake` .

I have found a workaorund by removing `libyaml ` from `DEPENDS`  of `rcl-yaml-param-parser`.

Maybe [https://github.com/ros2/rcl/issues/638](https://github.com/ros2/rcl/issues/638) is relayted to this problem?

beralt, 2021-08-20:
Having absolute paths in certain packages is something I'm running in to as well. I'm trying to build an SDK. After a number of changes, which I'm planning to offer as a PR, I can build an SDK that is capable of building ROS2 components. But these absolute paths still pop up and CMake correctly warns about them.

Most of the ROS2 packages have been fixed but some external dependencies still result in absolute paths. These include `libyaml` in `rcl-yaml-param-parser`. The fixes for these is to make `yaml` a CMake target using `add_library` for example.

For an OpenEmbedded build an absolute path in the CMake Config exports should never be allowed, the sysroot for the package is temporary and thus the path does not make any sense. I have modified the `ament-cmake-target-dependencies` to emit a warning when it is about to use an absolute path.

Any ideas about this? I'm not sure why upstream `ament-cmake` is using absolute paths. Any ROS2 package that has external dependencies and does not generate a CMake target for it is affected by this.

graceagrace, 2024-11-05:
> I have found a similar problem like this also with `rcl-yaml-param-parser`, in an sdk build `librcl `->`librcl_yaml_param_parser `has a hardcoded wrong path in `rcl_yaml_param_parser/cmake/rcl_yaml_param_parserExport.cmake` .
> 
> I have found a workaround by removing `libyaml ` from `DEPENDS` of `rcl-yaml-param-parser`.
> 
> Maybe [ros2/rcl#638](https://github.com/ros2/rcl/issues/638) is relayted to this problem?

I'm aware this is an old issue, but seeing as it's unresolved... I'm encountering the same problem now with kirkstone and humble. I removed `libyaml` from `ROS_BUILD_DEPENDS`, `ROS_EXPORT_DEPENDS`, and `ROS_EXEC_DEPENDS`, but am still running into the hardcoded wrong path issue. Did you have to do anything else to resolve this?

robwoolley, 2024-11-21:
Hi @graceagrace , I think this might be what you are looking for: https://github.com/ros/meta-ros/commit/4061ecd3327c6e6680254777a4f623efdcf412b9