ros/meta-ros #794: Having an issue integrating with the `meta-xilinx` and `meta-iveia` layers.

naterenegar, 2021-02-06:
Hello,

I'm trying to integrate `meta-ros`, specifically the (unfortunately unsupported) `thud` branch with these layers:

- [meta-xilinx](https://github.com/Xilinx/meta-xilinx/tree/rel-v2019.2)
- [meta-iveia](https://github.com/iVeia/meta-iveia)

I'm able to get the above to build cleanly using the steps on the meta-iveia page.

When I added in ROS using the [steps in the wiki](https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project), the python distribution generated by bitbake seems to fail when building `btrfs` because it cannot find `setuptools`:

```
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 16
|     [LD]     btrfs
|     [PY]     libbtrfsutil
| Traceback (most recent call last):
|   File "setup.py", line 23, in <module>
|     from setuptools import setup, Extension
| ModuleNotFoundError: No module named 'setuptools'
| Makefile:433: recipe for target 'libbtrfsutil_python' failed
| make: *** [libbtrfsutil_python] Error 1
| make: *** Waiting for unfinished jobs....
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile 
```

I've found a couple of people with similar issues, but the answers are either incomplete or not very helpful:

- https://answers.ros.org/question/343341/boost-and-btrfs-tools-failed-yocto-with-meta-ros-layer-build/
- https://stackoverflow.com/questions/60060832/boost-and-btrfs-tools-failed-with-yocto-build

Any help would be greatly appreciated!

naterenegar, 2021-02-06:
The source of the issue seems to be that `thud` uses python 3.5 but `meta-ros-backports-warrior` uses python 3.7. I tried setting the ``BBFILE_PRIORITY` of that layer to 2 so that 3.5 was the version used, but it created a bunch of quality assurance errors:
```
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-lic went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-tests went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-2to3 went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-asyncio went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-audio went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-codecs went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-compile went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-compression went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-core went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-crypt went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-ctypes went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-curses went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-datetime went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-db went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-debugger went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-difflib went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-distutils-staticdev went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-distutils went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-doctest went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-email went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-fcntl went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-gdbm went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-html went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-idle went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-image went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
[... cut ...]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-locale went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-misc went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
ERROR: python3-3.5.6-r1.0 do_packagedata: QA Issue: Package version for package python3-man went backwards which would break package feeds from (0:3.7.5-r0 to 0:3.5.6-r1.0) [version-going-backwards]
```

robwoolley, 2023-11-17:
Sorry that you didn't get a respond to your question.  I'm closing this since thud is unsupported and this issue is over 2 years old.  Please reopen if you have any follow-up questions.

closed 2023-11-17 by robwoolley