ros/meta-ros #799: Rviz can't be build in meta-ros1-melodic

minhthuc2502, 2021-02-18:
Hello,

I'm trying to use meta-ros on `branch zeus` for my card which run only version zeus of yocto. i used steps in [wiki].(https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project)

I want to run the simulation rviz on this card so i tried to use recipe `rviz` by `bitbake rviz`. The problem is that rviz uses Qt5 and it can't find the path for header files of Qt. As i checked, the directory of qt is installed correctly in `recipe-sysroot`. 

[image]

Before, I have some errors like missing `sip binding generator`, then i added `sip sip-native and python-sip` for rviz and let `PNBLACKLIST[rviz]` empty to unblacklist rviz.

What is the configuration I missed?

Any help would be greatly appreciated!

ABTrac, 2022-04-14:
Hi @minhthuc2502 and community 

I am running into a similar problem . I have added meta-ros to an existing image as instructed in [https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project](url) and everything was successfully compiled. However, when I wanted to add Rviz to the image I got the following configuration errors:

[image]

I am building the image from the Dunfell branch, and I have added `meta-qt5` layer and also added sip in local.conf  `IMAGE_INSTALL_append= " sip"` and removed rviz from the blacklist.  @minhthuc2502 I would really appreciate if you can please elaborate what you did to fix the  `SIP binding generator NOT available ` error. Thank you very much :)

robwoolley, 2024-04-10:
Closing as Melodic is no longer supported.

closed 2024-04-10 by robwoolley