ros/meta-ros #803: How to contol RWM  implementations of the target image?

hannes09, 2021-02-20:
I could not find any information in the wiki, how to contorl whitch rwm are included in the final image.

I am missing something like: Only depend on "fastrtps" or  only include "cyclone" in the image.
I dont want to have muliple rwm implementations in the image, as it only would waste space, and never is used.

Is it possible?

shr-project, 2021-06-03:
See https://github.com/ros/meta-ros/pull/854 with that both fastrtps and cyclonedds are installed in every rolling and galactic image built with meta-ros.

If you want to remove one of them, then you would need to add a bbappend for rmw-implementation and remove the one you don't want from ROS_EXEC_DEPENDS.

Similarly like "rmw-connext-cpp rmw-connextdds" is removed from ROS_BUILD_DEPENDS when connext is in ROS_WORLD_SKIP_GROUPS.

You can also send a patch to support fastrtps and cyclonedds as ROS_WORLD_SKIP_GROUPS to remove this more automatically based on ROS_WORLD_SKIP_GROUPS (especially if there are more recipes ending with runtime dependency on the implementation you want to remove)

robwoolley, 2024-11-21:
This appears to have been answered already.  Please reopen the issue if you have any follow-up questions.

closed 2024-11-21 by robwoolley