ros/meta-ros #811: ROS2 with Toradex Build a Reference Image with Yocto Project

ierturk, 2021-03-24:
Hi,

I'm newbie to build yocto, I followed the guid for `Build a Reference Image with Yocto Project` [1], I was succeeded. However, when I added `ROS2` layer into `bblayers.conf `as follow as described here[2] I got the error message below.

```
 ...
 ${TOPDIR}/../layers/openembedded-core/meta \
 ${TOPDIR}/../../extra/meta-ros/meta-ros2-foxy \
 ...
```
the error message
```
 bitbake tdx-reference-minimal-image
 ERROR: Unable to start bitbake server (None)
 ERROR: Server log for this session (/media/ierturk/Work/REPOs/toradex/yocto/oe_core/build/bitbake-cookerdaemon.log):
 --- Starting bitbake server pid 2720 at 2021-03-24 14:50:17.098520 ---
 ERROR: Layer ros2-foxy-layer is not compatible with the core layer which only supports these series: dunfell (layer is compatible with ${ROS_OE_RELEASE_SERIES})
```

Environment:

    * Ubuntu 20.04 - 64 bit
    * Target Board: Toradex Verdin Mini

Thanks in advance...

Ibrahim

[1] https://developer.toradex.com/knowledge-base/board-support-package/openembedded-core
[2] https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project

shr-project, 2021-03-24:
What meta-ros revision are you using? You're definitely missing bunch of layer dependencies like meta-ros-common (which sets ROS_OE_RELEASE_SERIES and also e.g. meta-ros2, meta-oe, meta-python etc)

ierturk, 2021-03-24:
Hi @shr-project,

It seems to be OK when following line added into `bblayers.conf` as you suggested.
```
  ${TOPDIR}/../../extra/meta-ros/meta-ros-common \
  ${TOPDIR}/../../extra/meta-ros/meta-ros-backports-gatesgarth \
  ${TOPDIR}/../../extra/meta-ros/meta-ros2 \
  ${TOPDIR}/../../extra/meta-ros/meta-ros2-foxy \
```
Thanks for your the reply. You saved my day. Building started successfully. Once the image will be created, I'll check it then let you know.

ierturk, 2021-03-26:
Hi @shr-project thanhs again. Everything is OK with the image ROS2 Foxy and Torizon Yocto Verdin.

closed 2021-03-26 by ierturk