ros/meta-ros #812: meta-ros2-foxy build fails on spdlog-vendor (missing: Patch_EXECUTABLE)

JackNWhite, 2021-03-26:
I am building a Xilinx Petalinux 2020.2 root filesystem with ROS2 (this is Yocto-based). The only user layers are the three from this repo: `meta-ros-common`, `meta-ros2` and `meta-ros2-foxy`. The branch is `zeus` - mandated by Petalinux.

I have added the recipes to the Yocto build by getting the list of recipes for these layers from the OpenEmbedded layer index and running them through this script:

```bash
function checkRecipe {
    if [ "$1" ]; then
        matching_recipes=$(find . -name $1 -type d | grep "ros2/\|ros2-foxy/\|ros-common/")

        if [ "$matching_recipes" != "" ]; then
            echo $1
        fi
    fi
}

for recipe in $@; do
    checkRecipe $recipe
done
```

Is anyone able to help with this, please? At the beginning of the build, I get this warning:
```
WARNING: No recipes available for:
  /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/project-spec/meta-ros/meta-ros-common/recipes-support/spdlog/spdlog_1.5.0.bbappend
```

However, I do have this recipe in:
```
./project-spec/meta-ros/meta-ros2-foxy/recipes-bbappends/spdlog-vendor
./project-spec/meta-ros/meta-ros2-foxy/generated-recipes/spdlog-vendor
```

The specific `log.do_configure` for `spdlog-vendor` is:
```
DEBUG: Executing shell function do_configure
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc
-- Check for working C compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++
-- Check for working CXX compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.6 (/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/ament_cmake/cmake)
-- Found PythonInterp: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.7.6", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
CMake Error at /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Patch (missing: Patch_EXECUTABLE)
Call Stack (most recent call first):
  /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPatch.cmake:68 (find_package_handle_standard_args)
  CMakeLists.txt:59 (find_package)
  CMakeLists.txt:83 (build_spdlog)

-- Configuring incomplete, errors occurred!
See also "/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/build/CMakeFiles/CMakeOutput.log".
WARNING: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/temp/run.do_configure.2533270:1 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DBUILD_TESTING=OFF -DPYTHON_SOABI=cpython-37m-aarch64-linux-gnu -Wno-dev'
ERROR: Execution of '/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/temp/run.do_configure.2533270' failed with exit code 1:
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc
-- Check for working C compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++
-- Check for working CXX compiler: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.9.6 (/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/ament_cmake/cmake)
-- Found PythonInterp: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.7.6", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/bin/python3-native/python3
CMake Error at /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Patch (missing: Patch_EXECUTABLE)
Call Stack (most recent call first):
  /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/FindPatch.cmake:68 (find_package_handle_standard_args)
  CMakeLists.txt:59 (find_package)
  CMakeLists.txt:83 (build_spdlog)

-- Configuring incomplete, errors occurred!
[... cut ...] H=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/home/jack/ws_zcu102/petalinux/2021_03_26_Ubuntu_Petalinux/build/tmp/work/aarch64-xilinx-linux/spdlog-vendor/1.1.2-1-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DBUILD_TESTING=OFF -DPYTHON_SOABI=cpython-37m-aarch64-linux-gnu -Wno-dev'
```

shr-project, 2021-03-26:
The first error is about spdlog recipe not spdlog-vendor and that's because you didn't include meta-ros-backport-dunfell layer, see:

```
Switched to branch 'zeus'
meta-ros$ find . -name spdlog\*
./meta-ros2-foxy/generated-recipes/spdlog-vendor
./meta-ros2-foxy/generated-recipes/spdlog-vendor/spdlog-vendor_1.1.2-1.bb
./meta-ros2-foxy/recipes-bbappends/spdlog-vendor
./meta-ros2-foxy/recipes-bbappends/spdlog-vendor/spdlog-vendor_1.1.2-1.bbappend
...
./meta-ros-backports-dunfell/recipes-support/spdlog
./meta-ros-backports-dunfell/recipes-support/spdlog/spdlog_1.5.0.bb
./meta-ros-common/recipes-support/spdlog
./meta-ros-common/recipes-support/spdlog/spdlog_1.5.0.bbappend
```

Once you add meta-ros-backports-dunfell layer it will probably build spdlog-vendor fine as well (as spdlog will be new enough).

Zeus is no longer supported, you should ask Xilinx to support at least Dunfell which is LTS release. But I'm not seeing it in our builds (not even with zeus branch), so it might be something specific to petalinux or just caused by this missing layer.

JackNWhite, 2021-03-29:
Thank you, very much!

closed 2021-03-29 by JackNWhite