ros/meta-ros #814: How can I disable fortran in my build

JackNWhite, 2021-03-31:
Hi,

I'm using meta-ros on my Petalinux build, which is based on Yocto zeus(1). I want to use **foxy**, so I am including the **meta-ros-common**, **meta-ros2**, **meta-ros2-foxy** and **meta-ros-backports-dunfell**.

(1) I have asked Xilinx to support **dunfell** in their next Petalinux release, but I am not hopeful.

Here is the list of recipes I have added:
```
CONFIG_amazon-kvs-sdk-cpp-kinesis-video-producer
CONFIG_ament-cmake
CONFIG_ament-cmake-ros
CONFIG_ament-lint
CONFIG_ament-nodl
CONFIG_ament-package
CONFIG_angles
CONFIG_apriltag
CONFIG_aws-cpp-sdk
CONFIG_bullet
CONFIG_cartographer
CONFIG_cartographer-ros
CONFIG_class-loader
CONFIG_collada-dom
CONFIG_common-interfaces
CONFIG_console-bridge
CONFIG_console-bridge-vendor
CONFIG_control-msgs
CONFIG_cppcheck
CONFIG_cppunit
CONFIG_cyclonedds
CONFIG_depthimage-to-laserscan
CONFIG_eigen-stl-containers
CONFIG_eigen3-cmake-module
CONFIG_example-interfaces
CONFIG_fastcdr
CONFIG_fastrtps
CONFIG_fcl
CONFIG_filters
CONFIG_fmilibrary
CONFIG_fmilibrary-vendor
CONFIG_foonathan-memory
CONFIG_foonathan-memory-vendor
CONFIG_geographic-info
CONFIG_geographiclib
CONFIG_geometry2
CONFIG_gps-umd
CONFIG_graphicsmagick
CONFIG_graphviz
CONFIG_ifm3d-core
CONFIG_image-common
CONFIG_image-pipeline
CONFIG_image-publisher
CONFIG_image-transport-plugins
CONFIG_interactive-markers
CONFIG_joint-state-publisher
CONFIG_joy
CONFIG_joystick
CONFIG_kdl-parser
CONFIG_laser-geometry
CONFIG_laser-proc
CONFIG_launch
CONFIG_launch-ros
CONFIG_lgsvl-msgs
CONFIG_libccd
CONFIG_libflann
CONFIG_libfreenect
CONFIG_libphidget22
CONFIG_libstatistics-collector
CONFIG_libyaml-vendor
CONFIG_log4cxx
CONFIG_message-filters
[... 309 lines cut ...]
checking whether to build static libraries... no
checking for aarch64-xilinx-linux-gfortran... aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot
checking whether we are using the GNU Fortran compiler... no
checking whether aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot accepts -g... no
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
libtool.m4: error: problem compiling FC test program
checking for aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot option to produce PIC... 
checking if aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot static flag  works... no
checking if aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot supports -c -o file.o... no
checking if aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot supports -c -o file.o... (cached) no
checking whether the aarch64-xilinx-linux-gfortran  -march=armv8-a+crc -mtune=cortex-a72.cortex-a53 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot linker (aarch64-xilinx-linux-ld --sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot ) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether the GNU Fortran compiler is working... no
configure: error: GNU Fortran is not working; please report a bug in http://gcc.gnu.org/bugzilla, attaching /home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/gcc-9.2.0/build.aarch64-xilinx-linux.aarch64-xilinx-linux/aarch64-xilinx-linux/libgfortran/config.log
WARNING: /home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/temp/run.do_configure.1683514:1 exit 1 from '../$relpath/$target/configure --build=x86_64-linux --host=aarch64-xilinx-linux --target=aarch64-xilinx-linux --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot --with-gnu-ld --enable-shared --enable-languages=c,c++,fortran --enable-threads=posix --disable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=aarch64-xilinx-linux- --without-local-prefix --enable-lto --disable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-sysroot=/not/exist --with-build-sysroot=/home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/recipe-sysroot --disable-static --disable-nls --with-glibc-version=2.28 --enable-initfini-array --enable-__cxa_atexit'

ERROR: Logfile of failure stored in: /home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/libgfortran/9.2.0-r0/temp/log.do_configure.1683514
```

Presumably, this means that the GNU Fortran compiler is disabled, and I can see from a post on the Xilinx forum that Xilinx says the Fortran compiler is "not supported", however, it is still in their GCC repo.

I can't work out from the above log what is being built that requires **gfortran**. Is there a way of easily finding this out? I am somewhat of a Yocto noob.

shr-project, 2021-03-31:
With git grep you can see:
```
meta-ros-common/conf/ros-distro/include/enable-fortran.inc:FORTRAN_forcevariable = "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'fortran', '', ',fortran', d)}"
meta-ros-common/conf/ros-distro/ros-distro.conf:# "lapack" needs FORTRAN support
```

the easiest option is to add fortran to ROS_WORLD_SKIP_GROUPS ti disable this.

JackNWhite, 2021-04-01:
Thank you; I decided to go the other way and get Fortran working in the build.

closed 2021-04-01 by JackNWhite