ros/meta-ros #815: Failure to retrieve osrf_testings_tools_cpp-release

JackNWhite, 2021-04-01:
### Description
When building `meta-ros` `zeus` as part of a Petalinux 2020.02 build, `osrf_testings_tools_cpp-release` consistently fails to be downloaded. This causes the Yocto build to fail.

### Reproduction
This is a barebones Petalinux 2020.02 build, with the addition of the Foxy layers from this repo:
```
core
meta-browser
meta-clang
meta-jupyter
meta-openamp
meta-openembedded
meta-petalinux
meta-qt5
meta-virtualization
meta-vitis-ai
meta-xilinx
meta-xilinx-tools
meta-ros-common
meta-ros2
meta-ros2-foxy
meta-ros-backports-dunfell
```

Additionally, Fortran is enabled in **project-spec/meta-user/conf/petalinuxbsp.conf**:
```
SIGGEN_UNLOCKED_RECIPES += "gcc-cross-aarch64 libgcc-initial"
FORTRAN_forcevariable = ",fortran"
```
The build is for the Xilinx ZCU102 board. The host OS is Linux Mint 20.1 (Ubuntu 20.04). 

### The Error
The detailed log is in [log.do_fetch.147654.txt](https://github.com/ros/meta-ros/files/6243728/log.do_fetch.147654.txt).

These lines appear in the build log:
```
WARNING: osrf-testing-tools-cpp-1.3.2-1-r0 do_fetch: Failed to fetch URL git://github.com/ros2-gbp/osrf_testings_tools_cpp-release;branch=release/foxy/osrf_testing_tools_cpp;protocol=https, attempting MIRRORS if available
ERROR: osrf-testing-tools-cpp-1.3.2-1-r0 do_fetch: Fetcher failure: Unable to find revision 343ee1608d087886d9e9a9a6188abb8129fdc37c in branch release/foxy/osrf_testing_tools_cpp even from upstream
ERROR: osrf-testing-tools-cpp-1.3.2-1-r0 do_fetch: Fetcher failure for URL: 'git://github.com/ros2-gbp/osrf_testings_tools_cpp-release;branch=release/foxy/osrf_testing_tools_cpp;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/jack/ws_zcu102/petalinux/Petalinux_Basic/build/tmp/work/aarch64-xilinx-linux/osrf-testing-tools-cpp/1.3.2-1-r0/temp/log.do_fetch.147654
ERROR: Task (/home/jack/ws_zcu102/petalinux/Petalinux_Basic/project-spec/meta-ros/meta-ros2-foxy/generated-recipes/osrf-testing-tools-cpp/osrf-testing-tools-cpp_1.3.2-1.bb:do_fetch) failed with exit code '1'
```

There is no branch `release/foxy/osrf_testing_tools_cpp` in `github.com/ros2-gbp/osrf_testings_tools_cpp-release`, but there is a branch `release/foxy/osrf_testing_tools_cpp/1.3.2-1`, however the former is referenced in the bbappend file for this recipe. The commit hash mentioned in the error _is_ in that branch.

**Expected behavior**
`osrf-testing-tools-cpp-1.3.2-1-r0` is downloaded correctly and builds.

shr-project, 2021-04-01:
see https://github.com/ros/meta-ros/commit/783087e018017c885815db19107ac28eeb4d2f15 in dunfell branch

JackNWhite, 2021-04-01:
Thank you once again!

closed 2021-04-07 by JackNWhite