ros/meta-ros #834: pcl-ros cannot compile for ROS noetic because of qhull update

lirenlin, 2021-04-16:
With this commit, the qhull version is updated in meta-ros-common.
https://github.com/ros/meta-ros/commit/86b7dabb19eb8a59a4cde0fabdddb354a11a8ac0

pcl_1.8.1 cannot compile properly with qhull_2020.2
The cmake module within pcl_1.8.1 is trying to find the library with old suffix, "qhull_p, qhull_pd".

In the new version of qhull_2020.2, it is renamed to "qhul_r".
pcl-ros depends on pcl which requires qhull to be present. 

The error is observed with dunfall branch. I would assume it should also affect the main.

shr-project, 2021-04-16:
Yes, there are many issues left in noetic, see last status update in:
https://github.com/ros/meta-ros/pull/828#issuecomment-818750205

Patches are welcome.

lirenlin, 2021-04-28:
which fix do you think is the best solution.
Locally, I copied the original old recipe into noetic folder and prefer the older version in the config, instead of the version in common.

shr-project, 2021-05-28:
> Locally, I copied the original old recipe into noetic folder and prefer the older version in the config, instead of the version in common.

`git mv ./meta-ros1-melodic/recipes-extended/qhull meta-ros1/recipes-extended/`

"fixes" the build issue for pcl-ros, but then
`meta-ros/meta-ros1-noetic/generated-recipes/geometric-shapes/geometric-shapes_0.7.3-1.bb:do_configure`
will fail again, so the best solution is to find a version which is compatible with both (or modify either qhull, or pcl-ros or geometic-shapes to be compatible)

I've added many fixes to https://github.com/ros/meta-ros/pull/849 , now only few recipes are failing for me (including either pcl-ros or geometric-shapes) so I'll probably get to this soon.

closed 2021-05-31 by shr-project