ros/meta-ros #845: Failed to build image(Zeus) with ROS1

alphaanyms05, 2021-05-10:
I am trying to build Zeus image including ROS1. Using these layers:

> meta-ros-common
> meta-ros1

I get build issues. Has anyone else had this problem?

Here is the error log:

**ERROR:** aws-checksums-0.1.5-r0 do_fetch: Fetcher failure: Unable to find revision 519d6d9093819b6cf89ffff589a27ef8f83d0f65 in branch master even from upstream
ERROR: aws-checksums-0.1.5-r0 do_fetch: Fetcher failure for URL: 'git://github.com/awslabs/aws-checksums.git;protocol=https'. Unable to fetch URL from any source.

**ERROR:** Logfile of failure stored in: /home/alpha/ros1_for_yocto_zeus/tmp/work/core2-64-oe-linux/aws-checksums/0.1.5-r0/temp/log.do_fetch.733201

**ERROR:** Task (/home/alpha/ros1_for_yocto_zeus/meta-ros/meta-ros-common/recipes-devtools/awslabs/aws-checksums_0.1.5.bb:do_fetch) failed with exit code '1'
WARNING: aws-c-common-0.4.15-r0 do_fetch: Failed to fetch URL git://github.com/awslabs/aws-c-common.git;protocol=https, attempting MIRRORS if available
ERROR: aws-c-common-0.4.15-r0 do_fetch: Fetcher failure: Unable to find revision e3e7ccd35a85f9cd38c67cb1988251f1543b6632 in branch master even from upstream

**ERROR:** aws-c-common-0.4.15-r0 do_fetch: Fetcher failure for URL: 'git://github.com/awslabs/aws-c-common.git;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/alpha/ros1_for_yocto_zeus/tmp/work/core2-64-oe-linux/aws-c-common/0.4.15-r0/temp/log.do_fetch.737671

**ERROR:** Task (/home/alpha/ros1_for_yocto_zeus/meta-ros/meta-ros-common/recipes-devtools/awslabs/aws-c-common_0.4.15.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4347 tasks of which 2868 didn't need to be rerun and 2 failed.

shr-project, 2021-05-10:
They renamed master branch to main in these repos, will send PR shortly, please test it.

alphaanyms05, 2021-05-13:
Thank you @shr-project for your quick response.
I solved the problem with the updated branch revision id

closed 2021-05-13 by alphaanyms05