ros/meta-ros #856: Nothing RPROVIDES 'nativesdk-catkin'

miram-jraidi, 2021-06-04:
i am trying to build a Linux image with meta-ros-dashing  layer branch dunfell but when I tried to bitbake ros-image-sdktest
this error always shows up 
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 4335 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'nativesdk-catkin'
NOTE: Runtime target 'nativesdk-catkin' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['nativesdk-catkin']
ERROR: Required build target 'ros-image-sdktest' has no buildable providers.
Missing or unbuildable dependency chain was: ['ros-image-sdktest', 'nativesdk-catkin']

hannes09, 2021-06-09:
Same error comes also with sdktest of ros-noetic.

I have addded `BCLASSEXTEND=“native nativesdk” ` to a catkin_%.bbappend

That will build the sdk.

To use th sdk I  sourced the setup.bash of the sdk like this:

`source my_sdk/x86_64-tdsdk-linux/opt/ros/noetic/setup.bash`

But unfortunataly I got:

`CMake Error at CMakeLists.txt:65 (message):
  find_package(catkin) failed.  catkin was neither found in the workspace nor
  in the CMAKE_PREFIX_PATH.  One reason may be that no ROS setup.sh was
  sourced before.`

Some question was answerd here:

[(https://stackoverflow.com/questions/64686050/yocto-sdk-ros1-melodic-meta-ros-build?answertab=votes#tab-top)]

Some somethings seems still broken.

hannes09, 2021-06-09:
I think this is a duplicate of [https://github.com/ros/meta-ros/issues/745](https://github.com/ros/meta-ros/issues/745)

robwoolley, 2024-11-21:
Closing as a duplicate.

closed 2024-11-21 by robwoolley