ros/meta-ros #861: ament_cmake Problem

jiexuwow, 2021-06-09:
1. I'm using Yocto to build my rootfs and which setup file should be used?
    /etc/profile.d/ros/local_setup.sh
    /usr/bin/ros_setup.sh
    /usr/bin/ros_local_setup.sh 
2. When I use `colcon build` to build package , the error is like this:
```
    
CMake Error at CMakeLists.txt:9 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< pkg [1.23s, exited with code 1]
```

    
How can I fix this ？ Thanks

Seajinbaek, 2023-09-15:
it can't fix

jsb9945, 2023-12-06:
Hello, How did you solve it?

Marshall-Wen-Amba, 2024-01-31:
Hi All:

Does anyone know how to resolve it properly? We also meet this issue when using the latest **meta-ros**(till 30th, Jan 2024) and **humble**.

// Below are the cmds that we used to build the "hello-world" package
`# mkdir /home/root/dev_ws
# cd /home/root/dev_ws
# ros2 pkg create --build-type ament_cmake --license Apache-2.0 --node-name my_node my_package
# colcon build`

`Starting >>> my_package
--- stderr: my_package
CMake Error at CMakeLists.txt:9 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.
 
  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:
 
    ament_cmakeConfig.cmake
    ament_cmake-config.cmake
 
  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.
 
 
---
Failed   <<< my_package [0.21s, exited with code 1]
 
Summary: 0 packages finished [0.81s]
  1 package failed: my_package
  1 package had stderr output: my_package
`

Marshall-Wen-Amba, 2024-01-31:
> it can't fix

Hi Seajinbaek:

Nice to meet you.
Do you mean if it is a known limitation of meta-ros? If yes, could you help to explain a bit more about the reason?

jsb9945, 2024-02-01:
Hi, @Marshall-Wen-Amba 
I don't know how to build using colcon, but I succeeded by creating a recipe and building it. Please give it a try.

Marshall-Wen-Amba, 2024-02-01:
Hi All:
Finally, I made it work by using "colcon build" to do native build on the board for both CMake and Python.
It is essentially about adding many PN-dev into IMAGE-INSTALL because we need header files and CMake configure files.

Enjoy~~

Marshall-Wen-Amba, 2024-02-01:
> Hi, @Marshall-Wen-Amba I don't know how to build using colcon, but I succeeded by creating a recipe and building it. Please give it a try.

Thanks for your response.

jsb9945, 2024-02-03:
@Marshall-Wen-Amba Congratulations. Can you tell me what the PN-dev package you mentioned is??

Marshall-Wen-Amba, 2024-02-04:
> @Marshall-Wen-Amba Congratulations. Can you tell me what the PN-dev package you mentioned is??

Sure. Here you go and enjoy~~

Add ROS2 dev packages into IMAGE_INSTALL. It will **copy cmake config files and header files** into rootfs.
````````
IMAGE_INSTALL:append = " \
......
                        python3-colcon-bash-dev \
                        python3-colcon-cd-dev \
                        python3-colcon-cmake-dev \
                        python3-colcon-common-extensions-dev \
                        python3-colcon-core-dev \
                        python3-colcon-defaults-dev \
                        python3-colcon-devtools-dev \
                        python3-colcon-library-path-dev \
                        python3-colcon-metadata-dev \
                        python3-colcon-output-dev \
......
``````

jsb9945, 2024-02-04:
@Marshall-Wen-Amba Thanks for your help. I added dev packages
but i got this error. I'd appreciate it if you could check it out.

`Starting >>> cpp_pubsub
[3.828s] colcon.colcon_ros.prefix_path.ament WARNING The path '/usr' in the environment variable AMENT_PREFIX_PATH doesn't contain any 'local_setup.*' files.
--- stderr: cpp_pubsub
Traceback (most recent call last):
  File "/usr/lib64/python3.9/site-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib64/python3.9/site-packages/colcon_core/task/__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib64/python3.9/site-packages/colcon_ros/task/ament_cmake/build.py", line 59, in build
    rc = await extension.build(
  File "/usr/lib64/python3.9/site-packages/colcon_cmake/task/cmake/build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "/usr/lib64/python3.9/site-packages/colcon_cmake/task/cmake/build.py", line 172, in _reconfigure
    raise RuntimeError("Could not find 'cmake' executable")
RuntimeError: Could not find 'cmake' executable
---
Failed   <<< cpp_pubsub [0.15s, exited with code 1]`

Marshall-Wen-Amba, 2024-02-04:
> @Marshall-Wen-Amba Thanks for your help. I added dev packages but i got this error. I'd appreciate it if you could check it out.
>
> ## Starting >>> cpp_pubsub
> [3.828s] [...]

Seems like you didn't add cmake. To make your life easier, you can add all below ones.
Note that the rootfs size will be near 2GB.
``````
			python3-colcon-bash-dev \
			python3-colcon-cd-dev \
			python3-colcon-cmake-dev \
			python3-colcon-common-extensions-dev \
			python3-colcon-core-dev \
			python3-colcon-defaults-dev \
			python3-colcon-devtools-dev \
			python3-colcon-library-path-dev \
			python3-colcon-metadata-dev \
			python3-colcon-output-dev \
			python3-colcon-package-information-dev \
			python3-colcon-package-selection-dev \
			python3-colcon-parallel-executor-dev \
			python3-colcon-pkg-config-dev \
			python3-colcon-python-setup-py-dev \
			python3-colcon-recursive-crawl-dev \
			python3-colcon-ros-dev \
			python3-colcon-test-result-dev \
			python3-distlib-dev \
			python3-empy-dev \
			python3-flake8-dev \
			python3-ifcfg-dev \
			python3-lark-parser-dev \
			python3-nose-yanc-dev \
			python3-pyassimp-dev \
			python3-pydocstyle-dev \
			python3-pygraphviz-dev \
			python3-pyproj-dev \
			python3-pytest-cov-dev \
			python3-pytest-mock-dev \
			python3-transforms3d-dev \
			tvm-dev \
			python-rosdep-data-dev \
			python3-rosdep-dev \
			python3-rosdistro-dev \
			python3-rospkg-dev \
			fmilibrary-dev \
			foonathan-memory-dev \
			python3-construct-dev \
			python3-whichcraft-dev \
			ament-cmake-auto-dev \
			ament-cmake-core-dev \
			ament-cmake-export-definitions-dev \
			ament-cmake-export-dependencies-dev \
			ament-cmake-export-include-directories-dev \
			ament-cmake-export-interfaces-dev \
			ament-cmake-export-libraries-dev \
			ament-cmake-export-link-flags-dev \
			ament-cmake-export-targets-dev \
			ament-cmake-gen-version-h-dev \
			ament-cmake-gmock-dev \
			ament-cmake-google-benchmark-dev \
			ament-cmake-gtest-dev \
			ament-cmake-include-directories-dev \
			ament-cmake-libraries-dev \
			ament-cmake-nose-dev \
			ament-cmake-pytest-dev \
			ament-cmake-python-dev \
			ament-cmake-target-dependencies-dev \
			ament-cmake-test-dev \
			ament-cmake-version-dev \
			ament-cmake-dev \
			ament-cmake-catch2-dev \
			ament-cmake-ros-dev \
			ament-download-dev \
			ament-index-cpp-dev \
			ament-index-python-dev \
			ament-cmake-copyright-dev \
			ament-cmake-cppcheck-dev \
			ament-cmake-cpplint-dev \
			ament-cmake-flake8-dev \
			ament-cmake-lint-cmake-dev \
			ament-cmake-mypy-dev \
			ament-cmake-pclint-dev \
			ament-cmake-pep257-dev \
			ament-cmake-pycodestyle-dev \
			ament-cmake-pyflakes-dev \
			ament-cmake-xmllint-dev \
			ament-copyright-dev \
			ament-cppcheck-dev \
			ament-cpplint-dev \
			ament-flake8-dev \
			ament-lint-auto-dev \
			ament-mypy-dev \
			ament-pclint-dev \
			ament-pep257-dev \
			ament-pyflakes-dev \
			ament-xmllint-dev \
			ament-nodl-dev \
			ament-package-dev \
			std-msgs-dev \
			std-srvs-dev \
			topic-monitor-dev \
			topic-statistics-demo-dev \
			examples-rclcpp-minimal-publisher-dev \
			examples-rclcpp-minimal-subscriber-dev \
			fastrtps-dev \
			python-cmake-module-dev \
			rcl-action-dev \
			rcl-lifecycle-dev \
			rcl-yaml-param-parser-dev \
			rcl-dev \
			action-msgs-dev \
			lifecycle-msgs-dev \
			rcl-interfaces-dev \
			rcl-logging-interface-dev \
			rcl-logging-noop-dev \
			rcl-logging-spdlog-dev \
			rclcpp-action-dev \
			rclcpp-components-dev \
			rclcpp-lifecycle-dev \
			rclcpp-dev \
			rclpy-dev \
			rcpputils-dev \
			rcutils-dev \
			ros-environment-dev \
			ros-workspace-dev \
			ros2-control-dev \
			ros2topic-dev \
			rosidl-adapter-dev \
			rosidl-cli-dev \
			rosidl-cmake-dev \
			rosidl-generator-c-dev \
			rosidl-generator-cpp-dev \
			rosidl-parser-dev \
			rosidl-runtime-c-dev \
			rosidl-runtime-cpp-dev \
			rosidl-typesupport-interface-dev \
			rosidl-typesupport-introspection-c-dev \
			rosidl-typesupport-introspection-cpp-dev \
			rosidl-generator-dds-idl-dev \
			rosidl-default-generators-dev \
			rosidl-default-runtime-dev \
			rosidl-generator-py-dev \
			rosidl-runtime-py-dev \
			rosidl-typesupport-c-dev \
			rosidl-typesupport-cpp-dev \
			fastrtps-cmake-module-dev \
			rosidl-typesupport-fastrtps-c-dev \
			rosidl-typesupport-fastrtps-cpp-dev \
			topic-tools-interfaces-dev \
			topic-tools-dev \
			ros-base-dev \
			ros-core-dev \
`````

jsb9945, 2024-02-04:
@Marshall-Wen-Amba Thank you. I tried adding what you gave me, but the same error occurred
I just delete 'python3-construct-dev, python3-whichcraft-dev, ament-cmake-gen-version-h-dev, topic-tools-interfaces-dev, topic-tools-dev'.
Will this be a problem?

Marshall-Wen-Amba, 2024-02-04:
> @Marshall-Wen-Amba Thank you. I tried adding what you gave me, but the same error occurred I just delete 'python3-construct-dev, python3-whichcraft-dev, ament-cmake-gen-version-h-dev, topic-tools-interfaces-dev, topic-tools-dev'. Will this be a problem?

Should be OK as I think. 
Some simple questions:
1. Are you using the latest code of meta-ros?
2. Which Poky are you using? Kirkstone, right?
3. Which ROS2 are you using? humble?

For your info, I'm using latest meta-ros(Kirkstone branch) with Poky Kirkstone. And I'm using Humble.

jsb9945, 2024-02-04:
@Marshall-Wen-Amba 
Currently I'm not using the latest meta-ros. I'm using hardknott and galactic.

Marshall-Wen-Amba, 2024-02-04:
> @Marshall-Wen-Amba Currently I'm not using the latest meta-ros. I'm using hardknott and galactic.

You can take a quick try with it(for comparison). Should be quite smooth.

Marshall-Wen-Amba, 2024-02-04:
> @Marshall-Wen-Amba Currently I'm not using the latest meta-ros. I'm using hardknott and galactic.

One more comment: pls also add below ones into your IMAGE_INTALL. Both PN and PN-dev. Let me know if it still doesn't help.

``````
			python3-colcon-bash \
			python3-colcon-cd \
			python3-colcon-cmake \
			python3-colcon-common-extensions \
			python3-colcon-core \
			python3-colcon-defaults \
			python3-colcon-devtools \
			python3-colcon-library-path \
			python3-colcon-metadata \
			python3-colcon-output \
			python3-colcon-package-information \
			python3-colcon-package-selection \
			python3-colcon-parallel-executor \
			python3-colcon-pkg-config \
			python3-colcon-python-setup-py \
			python3-colcon-recursive-crawl \
			python3-colcon-ros \
			python3-colcon-test-result \
			python3-distlib \
			python3-empy \
			python3-flake8 \
			python3-ifcfg \
			python3-lark-parser \
			python3-nose-yanc \
			python3-pyassimp \
			python3-pydocstyle \
			python3-pygraphviz \
			python3-pyproj \
			python3-pytest-cov \
			python3-pytest-mock \
			python3-transforms3d \
			tvm \
			python-rosdep-data \
			python3-rosdep \
			python3-rosdistro \
			python3-rospkg \
			fmilibrary \
			foonathan-memory \
			python3-construct \
			python3-whichcraft \
			ament-cmake-auto \
			ament-cmake-core \
			ament-cmake-export-definitions \
			ament-cmake-export-dependencies \
			ament-cmake-export-include-directories \
			ament-cmake-export-interfaces \
			ament-cmake-export-libraries \
			ament-cmake-export-link-flags \
			ament-cmake-export-targets \
			ament-cmake-gen-version-h \
			ament-cmake-gmock \
			ament-cmake-google-benchmark \
			ament-cmake-gtest \
			ament-cmake-include-directories \
			ament-cmake-libraries \
			ament-cmake-nose \
			ament-cmake-pytest \
			ament-cmake-python \
			ament-cmake-target-dependencies \
			ament-cmake-test \
			ament-cmake-version \
			ament-cmake \
			ament-cmake-catch2 \
			ament-cmake-ros \
			ament-download \
			ament-index-cpp \
			ament-index-python \
			ament-cmake-copyright \
			ament-cmake-cppcheck \
			ament-cmake-cpplint \
			ament-cmake-flake8 \
			ament-cmake-lint-cmake \
			ament-cmake-mypy \
			ament-cmake-pclint \
			ament-cmake-pep257 \
			ament-cmake-pycodestyle \
			ament-cmake-pyflakes \
			ament-cmake-xmllint \
			ament-copyright \
			ament-cppcheck \
			ament-cpplint \
			ament-flake8 \
			ament-lint-auto \
			ament-mypy \
			ament-pclint \
			ament-pep257 \
			ament-pyflakes \
			ament-xmllint \
			ament-nodl \
			ament-package \
			std-msgs \
			std-srvs \
			topic-monitor \
			topic-statistics-demo \
			examples-rclcpp-minimal-publisher \
			examples-rclcpp-minimal-subscriber \
			fastrtps \
			python-cmake-module \
			rcl-action \
			rcl-lifecycle \
			rcl-yaml-param-parser \
			rcl \
			action-msgs \
			lifecycle-msgs \
			rcl-interfaces \
			rcl-logging-interface \
			rcl-logging-noop \
			rcl-logging-spdlog \
			rclcpp-action \
			rclcpp-components \
			rclcpp-lifecycle \
			rclcpp \
			rclpy \
			rcpputils \
			rcutils \
			ros-environment \
			ros-workspace \
			ros2-control \
			ros2topic \
			rosidl-adapter \
			rosidl-cli \
			rosidl-cmake \
			rosidl-generator-c \
			rosidl-generator-cpp \
			rosidl-parser \
			rosidl-runtime-c \
			rosidl-runtime-cpp \
			rosidl-typesupport-interface \
			rosidl-typesupport-introspection-c \
			rosidl-typesupport-introspection-cpp \
			rosidl-generator-dds-idl \
			rosidl-default-generators \
			rosidl-default-runtime \
			rosidl-generator-py \
			rosidl-runtime-py \
			rosidl-typesupport-c \
			rosidl-typesupport-cpp \
			fastrtps-cmake-module \
			rosidl-typesupport-fastrtps-c \
			rosidl-typesupport-fastrtps-cpp \
			topic-tools-interfaces \
			topic-tools \
			ros-base \
			ros-core \
```````

jsb9945, 2024-02-04:
@Marshall-Wen-Amba 
I still get the same error. I'll try the other way too. Thank you.

robwoolley, 2024-11-21:
This should be fixed in the supported Yocto releases (kirkstone, scarthgap, styhead, and walnascar (master)).

Most of the testing has been done on master+rolling.  It needs to be tested in the other combos, but it should work.  Please open a new issue if you encounter any problems and have follow up questions.

closed 2024-11-21 by robwoolley