ros/meta-ros #867: How to downgrade the yocto version with meta-ros

Justin790126, 2021-06-12:
Here I have to use yocto 3.1.0 to build meta-ros1-melodic for some usage.

What files should I need to edit to fetch the correspond yocto version with your SOP in instruction ?

shr-project, 2021-07-16:
What you mean? You cannot use newer dunfell branch revision than the initial 3.1.0? Why? 3.1.0 is missing many important changes and fixes.

Justin790126, 2021-07-23:
Because my hardware i500’s newest yocto version is yocto3.1.0. I have no choice but to use this version to build meta-ros.

shr-project, 2021-07-23:
Sorry, I don't know what i500 is or why it would be incompatible with 3.1.10 (latest revision of dunfell release as in https://lists.yoctoproject.org/g/poky/message/12511).

robwoolley, 2023-11-17:
This doesn't appear to be a bug and is over 2 years old.

closed 2023-11-17 by robwoolley