ros/meta-ros #868: pcl_1.8.1.bb:do_compile failed

yocto2222, 2021-06-15:
i am building  a image with meta ros foxy dunfell branch and i keep getting this error 
ERROR: pcl-1.8.1-r0 do_compile: Execution of '/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/temp/run.do_compile.8715' failed with exit code 1:
[1/313] /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DPCLAPI_EXPORTS -I/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/recognition/include/pcl/recognition/3rdparty -Iinclude -I/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/common/include -I/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/search/include -I/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include -isystem /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot/usr/include/eigen3 -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0=/usr/src/debug/pcl/1.8.1-r0                      -fdebug-prefix-map=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0=/usr/src/debug/pcl/1.8.1-r0                      -fdebug-prefix-map=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -ffloat-store -std=gnu++14  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/recipe-sysroot -pthread -fopenmp -O2 -g -DNDEBUG -fPIC   -DBOOST_DISABLE_ASSERTS -DEIGEN_NO_DEBUG -MD -MT sample_consensus/CMakeFiles/pcl_sample_consensus.dir/src/sac_model_normal_sphere.cpp.o -MF sample_consensus/CMakeFiles/pcl_sample_consensus.dir/src/sac_model_normal_sphere.cpp.o.d -o sample_consensus/CMakeFiles/pcl_sample_consensus.dir/src/sac_model_normal_sphere.cpp.o -c /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/src/sac_model_normal_sphere.cpp
In file included from /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include/pcl/sample_consensus/sac_model.h:52,
                 from /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include/pcl/sample_consensus/sac_model_normal_sphere.h:44,
                 from /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include/pcl/sample_consensus/impl/sac_model_normal_sphere.hpp:44,
                 from /media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/src/sac_model_normal_sphere.cpp:39:
/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include/pcl/sample_consensus/model_types.h: In function 'void __static_initialization_and_destruction_0(int, int)':
/media/sofia/mir/ros2/buildsofia/tmp/work/aarch64-poky-linux/pcl/1.8.1-r0/git/sample_consensus/include/pcl/sample_consensus/model_types.h:99:3: warning: 'pcl::SAC_SAMPLE_SIZE' is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   99 |   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
      |   ^~~~~~~~~~~~~~~
if there any way to fix it please help

shr-project, 2021-06-16:
The part of the log you've shared doesn't show the error, but most likely it's running out of memory. Can you add more ram to your builder or at least more swap?

schroettinger, 2022-12-22:
When it comes to pcl package I faced also issues with the build due to maxed out RAM consumption.

My best practice in order to solve this problem:
Rerun it two times, then it gets build properly.
Otherwise I guess you have to add some RAM as shr-project already pointed out.

(My setup: 8threads, 24GB RAM, dunfell 3.1.8, ROS2 galactic)

robwoolley, 2023-11-17:
I agree with the assessment that the error is likely due to the OOM killer.  This is over 2 years old and against foxy which is no longer supported.  Closing it as likely resolved given the comments above.

closed 2023-11-17 by robwoolley