ros/meta-ros #869: How to explicitly set python site packages path

Justin790126, 2021-06-17:
gencpp-0.6.5-1-r0 do_configure: Execution of '/home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/temp/run.do_configure.11423' failed with exit code 1:
-- The C compiler identification is Clang 10.0.0
-- The CXX compiler identification is Clang 10.0.0
-- Check for working C compiler: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
-- Check for working C compiler: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang++
-- Check for working CXX compiler: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/devel
-- Using CMAKE_PREFIX_PATH: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot/opt/ros/melodic;/home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot/usr
-- Found PythonInterp: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/python-native/python (found suitable version "2.7.18", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/python-native/python
-- Using default Python package layout
-- Found PY_em: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc  
-- Using empy: /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: 0
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
Traceback (most recent call last):
  File "/home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/build/catkin_generated/generate_cached_setup.py", line 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):

As describe above, ImportError: No module named catkin.environment_cache. However, I can see the file in the following path

 /home/justin126/mtk-git/build/tmp/work/aarch64-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/opt/ros/melodic/lib/python2.7/site-packages/catkin/environment_cache.py

I want to explicitly set the file in recipe(gencpp*.bb", How can I achieve it ?

shr-project, 2021-07-16:
Works for me.

Can you share some description of your build (as the Issue ticket template suggests?) from the output I've guessed that you're using poky DISTRO and some aarch64 MACHINE and are building probably melodic ROS distribution (based on catkin version shown in output). But there is nothing to guess what OE release you're using and what other layers are included (meta-clang maybe?)

And I've just built gencpp with poky/hardknott/qemuarm64 and meta-ros1-melodic without this issue.

shr-project, 2021-07-16:
```
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/jenkins/mjansa/build/poky/meta-python2/recipes-devtools/python/python-native_2.7.18.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta-ros/meta-ros1-melodic/generated-recipes/genmsg/genmsg_0.5.16-1.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/jenkins/mjansa/build/poky/meta-ros/meta-ros1-melodic/generated-recipes/catkin/catkin_0.7.29-1.bb:do_populate_sysroot', 'virtual:native:/jenkins/mjansa/build/poky/meta/recipes-devtools/ninja/ninja_1.10.2.bb:do_populate_sysroot', 'virtual:native:/jenkins/mjansa/build/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta/recipes-core/glibc/glibc_2.33.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta-python2/recipes-devtools/python/python_2.7.18.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta/recipes-devtools/cmake/cmake-native_3.19.5.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta/recipes-devtools/gcc/gcc-runtime_10.2.bb:do_populate_sysroot', 'virtual:native:/jenkins/mjansa/build/poky/meta-python2/recipes-devtools/python/python-setuptools_42.0.2.bb:do_populate_sysroot', 'virtual:native:/jenkins/mjansa/build/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/jenkins/mjansa/build/poky/meta/recipes-devtools/gcc/gcc-cross_10.2.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['python-native', 'genmsg', 'quilt-native', 'catkin-native', 'ninja-native', 'pseudo-native', 'glibc', 'python', 'cmake-native', 'gcc-runtime', 'python-setuptools-native', 'patch-native', 'gcc-cross-aarch64', 'gdbm-native', 'pkgconfig-native', 'xz-native', 'autoconf-native', 'bzip2-native', 'readline-native', 'automake-native', 'db-native', 'openssl-native', 'expat-native', 'sqlite3-native', 'zlib-native', 'libtool-native', 're2c-native', 'linux-libc-headers', 'openssl', 'opkg-utils', 'gdbm', 'libxcrypt', 'sqlite3', 'bzip2', 'libffi', 'zlib', 'db', 'readline', 'curl-native', 'ncurses-native', 'attr-native', 'catkin', 'python-catkin-pkg-native', 'python-nose-native', 'googletest-native', 'python-empy-native', 'libgcc', 'unzip-native', 'texinfo-dummy-native', 'flex-native', 'binutils-cross-aarch64', 'libmpc-native', 'mpfr-native', 'gnu-config-native', 'gmp-native', 'gettext-minimal-native', 'm4-native', 'ncurses', 'python-empy', 'python-catkin-pkg', 'python-nose', 'googletest', 'python-pyparsing-native', 'python-pyparsing']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing shell function do_configure
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/devel
-- Using CMAKE_PREFIX_PATH: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot/opt/ros/melodic;/jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot/usr
-- This workspace overlays: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot/opt/ros/melodic
-- Found PythonInterp: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/python-native/python (found suitable version "2.7.18", minimum required is "2")
-- Using PYTHON_EXECUTABLE: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/bin/python-native/python
-- Using default Python package layout
-- Found PY_em: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
-- Using empy: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/recipe-sysroot-native/usr/lib/python2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: 0
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- Using these message generators:
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CMAKE_INSTALL_BINDIR
    CMAKE_INSTALL_DATAROOTDIR
    CMAKE_INSTALL_INCLUDEDIR
    CMAKE_INSTALL_LIBDIR
    CMAKE_INSTALL_LIBEXECDIR
    CMAKE_INSTALL_LOCALSTATEDIR
    CMAKE_INSTALL_SBINDIR
    CMAKE_INSTALL_SHAREDSTATEDIR
    CMAKE_INSTALL_SYSCONFDIR
    CMAKE_NO_SYSTEM_FROM_IMPORTED
    LIB_SUFFIX
    PYTHON_EXECUTABLE_TARGET
    Python3_EXECUTABLE
    Python_EXECUTABLE

-- Build files have been written to: /jenkins/mjansa/build/poky/build/tmp/work/cortexa57-poky-linux/gencpp/0.6.5-1-r0/build
DEBUG: Shell function do_configure finished
DEBUG: Executing python function do_qa_configure
DEBUG: Python function do_qa_configure finished
```

Justin790126, 2021-07-23:
Hello, I had tested meta-ros on army arm64 in dunfell3.1.7, and it works perfect. My SoC is mediatek i500 (dunfell3.1.0). So far, I had find the root cause, the meta-ros I ported can not find out the dependencies in .bb file. The message in this issue can be solved by add ROS_EXPORT_DEPENDS += “ catkin” in .bb file. However, when it comes to build rostime, I can NOT solve the similar issue by add cpp-common to ROS_EXPORT_DEPENDS. May I have your personal Gmail to share the code base with you?

shr-project, 2021-07-23:
I would recommend you to compare the WORKDIRs of these components between the qemuarm64 and the i500 build to figure out what the mediatek BSP is doing wrong and they should definitely update to latest dunfell revision as well 3.1.0 doesn't even build on newer host distros.

Justin790126, 2021-07-27:
Do any recipes for testing meta-ros compilation? **ros-image-sdktest.bb**  is what I'm looking for ?

shr-project, 2021-08-23:
ros-image-world contains all generated recipes of selected ROS Distribution.

robwoolley, 2023-11-17:
This appears to have been resolved.  Closing it as it is greater than 2 years old.

closed 2023-11-17 by robwoolley