ros/meta-ros #876: Error in using python3

KirollosTadros, 2021-06-28:
I am using arago project of ti on the ti board TDA2Px-EVM there is meta-ros layer in this yocto project this layer is a clone form the legacy branch I was able to add that layer to the project and build the yocto image successfully without any errors, and I was also able to boot the board using that image and found that ros is installed.
I also exported the variables as described in the Readme but when I start roscore I got an error that no module called xmlrpc as if it running from python2 although the one installed inside the lib folder in ROS is python3.5.

I want to know how to handle that issue??

Thank you,
Best Regards,
Kirollos Henry

gorleabhinav, 2021-08-18:
hi, 
I did not find any python implementation for xmlrpc in the meta-ros layer, that's why your getting no module error.
If you want to include python library for xmlrp include the recipe for [aiohttp-xmlrpc](https://github.com/gorleabhinav/yocto/tree/main/aiohttp-xmlrpc).

KirollosTadros, 2021-08-18:
Hello,

Thanks for your reply.

Actually I manged to the solve the problem more than month ago.

The problem was that ROS is installed for python3 and and the system is yocto system that is having python2.7 and python3.5 but the system default was python3.

After some debugging inside roscore python codes if found that ROS was trying to run using python3 and everything related to ROS was installed in python3 so I have used update-alternatives command to solve that problem and make the default python of the system is python3

Thank you.

Best Regards,
Kirollos Henry

herb-kuta-lge, 2021-08-18:
> Actually I manged to the solve the problem more than month ago.

Closing this issue. Feel free to re-open if there's something that's still a problem.

closed 2021-08-18 by herb-kuta-lge