ros/meta-ros #885: meta-ros2-galactic shared libarries not present on the system

romi2002, 2021-07-12:
I've been trying to get meta-ros-galactic on a dunfell yocto confing for a Beaglebone Black board. The system boots fine, however, when I run a command such as `ros2 topic list`, I get the following error message: 

``` root@beaglebone:~# ros2 topic list
Failed to load entry point 'list': No module named 'rclpy._rclpy_pybind11'
The C extension '/usr/lib/python3.8/site-packages/_rclpy_pybind11.cpython-38-arm-linux-gnueabi.so' isn't present on the system.
```

The file `_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so` exists in `/usr/lib/python3.8/site-packages/rclpy`, and running `file` on it tells me it's compiled for ARM, even if the filename shows `x86_64`. Is meta-ros not naming these files correctly?

romi2002, 2021-07-12:
This seems to be an issue only for meta-ros-galactic, as when testing the same config for dashing, the issue does not happen.

shr-project, 2021-07-14:
Should be fixed by https://github.com/ros/meta-ros/pull/887:

```
Robot Operating System 2 (ROS 2) with OpenEmbedded Linux 2.galactic.2021-05-23.2-16+3.3.1-102-hardknott qemuarm ttyAMA0

qemuarm login: root
root@qemuarm:~# source ros_setup.sh
root@qemuarm:~# ros2 topic list
/parameter_events
/rosout
root@qemuarm:~# ros2 wtf
/usr/lib/python3.9/site-packages/ros2doctor/api/__init__.py: 112: UserWarning: Fail to call PackageCheck class functions.
/usr/lib/python3.9/site-packages/ros2doctor/api/__init__.py: 112: UserWarning: Fail to call PlatformCheck class functions.
1626265085.515484 [0]    python3: using network interface eth0 (udp/192.168.7.2) selected arbitrarily from: eth0, eth0

All 2 checks passed
```

closed 2021-07-14 by shr-project