ros/meta-ros #898: catkin.environment_cache error during bitbake

suyash-narain, 2021-07-31:
i am trying to add the meta-ros layer to my hardknott yocto image which i have created. 
after running bitbake ros-core, i get the following errors:

Traceback (most recent call last):
  File "/home1/usrdir/mtk23610/ros_images/nn/tmp/work/cortexa73-cortexa53-poky-linux/cmake-modules/0.4.2-r0/build/catkin_generated/generate_cached_setup.py", line 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /home1/usrdir/mtk23610/ros_images/nn/tmp/work/cortexa73-cortexa53-poky-linux/cmake-modules/0.4.2-r0/recipe-sysroot-native/opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  

Summary: 14 tasks failed:
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/cmake-modules/cmake-modules_0.4.2.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros/rosunit_1.14.9-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-comm/rosmaster_1.14.11-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-comm/rosgraph_1.14.11-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros/rosmake_1.14.9-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-comm/rosmsg_1.14.11-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-comm/roslaunch_1.14.11-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros/rosclean_1.14.9-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros/roscreate_1.14.9-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/metapackages/ros-core_1.4.1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/rosbag-migration-rule/rosbag-migration-rule_1.0.0.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros/rosboost-cfg_1.14.9-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-comm/rosparam_1.14.11-1.bb:do_configure
  /home1/usrdir/mtk23610/ros_images/src/poky/../meta-ros/meta-ros1-melodic/generated-recipes/ros-environment/ros-environment_1.2.3-1.bb:do_configure
Summary: There were 14 ERROR messages shown, returning a non-zero exit code.

what changes should i make in my local.conf to add catkin support and build the ros layer on top of my yocto layer?

shr-project, 2021-08-23:
Similar issue was mentioned in https://github.com/ros/meta-ros/issues/869 I recommend to follow https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions do the build in this setup (which will match with what we're regularly testing) and then compare the log.do_configure logs from this build with your setup to see what's different.

You didn't mention what other layers or configuration changes you added in your setup, so it's very difficult to guess what went wrong for you, but everything builds fine here and is built before any PR is merged.

suyash-narain, 2021-09-14:
it was an issue with the custom layer. I was able to resolve it. thank you

closed 2021-09-14 by suyash-narain