ros/meta-ros #905: fail to add GStreamer Plugin (kvssink)

gorleabhinav, 2021-08-17:
Trying to build the Amazon Kinesis Video Streams CPP Producer with GStreamer Plugin (kvssink).

Below line i have added in the amazon-kvs-sdk-cpp-1.7.8.inc file
**EXTRA_OECMAKE = "-DBUILD_GSTREAMER_PLUGIN=TRUE\"**

with the above changes able to build the Amazon Kinesis Video Streams CPP Producer, but did not get **libgstkvssink.so** binary file with package.

shr-project, 2021-09-10:
I've added recipes for amazon-kvs only to unblock building kinesis-manager, if you want some extra features from it, then you'll probably need to debug it yourself (unless someone in community is interested in the same).

robwoolley, 2024-11-21:
The aws recipes have now been removed.  Closing this issue.

closed 2024-11-21 by robwoolley