ros/meta-ros #919: Error : 'ROS_BUILD_TYPE:pn-python-orocos-kdl = "catkin"'

Kirollos-Henry, 2021-09-02:
Hello all,

I am using ti processor SDK it is based on yocto dunfell, I was able to customize it and added more than one layer like meta-scipy and I was also able to install some packages but when I added meta-ros1-noetic layer to my bitbtake project I got that error:

--- Starting bitbake server pid 16215 at 2021-09-02 13:46:21.635472 ---
ERROR: ParseError at /home/Kirollos/proc-sdk/tisdk/sources/meta-ros/meta-ros1-noetic/conf/ros-distro/include/noetic/ros-distro.inc:228: unparsed line: 'ROS_BUILD_TYPE:pn-python-orocos-kdl = "catkin"''

I was using an older version of processor SDK and it was based on thud I was working with the meta-ros legacy branch it was working just fine.

Now I am currently using texas instruments processor SDK that is an openembedded project base on yocto dunfell, and all I did is adding meta-ros1-noetic layer to the project now I can no longer run my bitbake server due to the parse error mention I hope this is a common issue and there is solution for it.

Best regards,
Kirollos Henry

shr-project, 2021-09-02:
You're probably missing recent commits from bitbake 1.46 branch:
https://git.openembedded.org/bitbake/commit/?h=1.46&id=a6d5fb7554e3cf071e453db56a1e7469ac44277c
https://git.openembedded.org/bitbake/commit/?h=1.46&id=c5418eae56cc50dbae7951c055434a0c456c53a4

Kirollos-Henry, 2021-09-02:
That solution solved my problem.
Thank you so much that was so helpful for me I was able to make bitbake server run that way.

I have one small question if I want to add ros full image to my recipe should I just add "packagegroup-ros-world-noetic" or what?

after adding this I got many dependencies error and I am moving from one to another is there and group that builds everything with is dependencies  

Thank you so much for your help

shr-project, 2021-09-02:
packagegroup-ros-world-noetic contains all packages generated for Noetic distribution (and should pull all necessary dependencies as well). It's what our CI is building regularly as this packagegroup is included in ros-image-world images.

If it fails for you with dependency issues, then there is something wrong in your setup (or just combination of the DISTRO_FEATURES and included layers you're using is significantly different from what is described in https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions and it's nearly impossible to test all possible combinations of ROS_WORLD_SKIP_GROUPS values).

Do you really want _all_ available packages? packagegroup-ros-world-noetic is good for CI, but for normal product you should include only what you really need in the image.

Kirollos-Henry, 2021-09-02:
Thank you so much for your help.
so you suggest that I have to build just core image and add the required packages after that?

The error I got on adding packagegroup-ros-world-noetic

ERROR: Nothing PROVIDES 'drone-wrapper' (but /home/Kirollos/proc-sdk/tisdk/sources/meta-ros/meta-ros1-noetic/generated-recipes/jderobot-drones/rqt-drone-teleop_1.4.0-1.bb DEPENDS on or otherwise requires it)
drone-wrapper was skipped: Recipe is blacklisted: Depends on gazebo, turtlebot3-gazebo which is not available
NOTE: Runtime target 'rqt-drone-teleop' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['rqt-drone-teleop', 'drone-wrapper']

Thank You so much for your help

shr-project, 2021-09-02:
Yes, this is the example of many possible combinations of ROS_WORLD_SKIP_GROUPS
```
meta-ros1-melodic/conf/ros-distro/include/melodic/ros-distro-recipe-blacklist.inc:PNBLACKLIST[rqt-drone-teleop] ?= "${@bb.utils.contains_any('ROS_WORLD_SKIP_GROUPS', ['qt5', 'pyqt5'], 'Requires rqt-gui->qt-gui->qtbase which requires meta-qt5 to be included and rqt-gui-py->qt-gui->python-pyqt5 which requires pyqt5', '', d)}"
meta-ros1-noetic/conf/ros-distro/include/noetic/ros-distro-recipe-blacklist.inc:PNBLACKLIST[rqt-drone-teleop] ?= "${@bb.utils.contains_any('ROS_WORLD_SKIP_GROUPS', ['qt5', 'pyqt5'], 'Requires rqt-gui->qt-gui->qtbase which requires meta-qt5 to be included and rqt-gui-py->qt-gui->python-pyqt5 which requires pyqt5', '', d)}"
```

These 2 lines should check for gazebo as well, because drone-wrapper is restricted with it:
```
meta-ros1-melodic/conf/ros-distro/include/melodic/ros-distro-recipe-blacklist.inc:PNBLACKLIST[drone-wrapper] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'gazebo', 'Depends on gazebo, turtlebot3-gazebo which is not available', '', d)}"
meta-ros1-noetic/conf/ros-distro/include/noetic/ros-distro-recipe-blacklist.inc:PNBLACKLIST[drone-wrapper] ?= "${@bb.utils.contains('ROS_WORLD_SKIP_GROUPS', 'gazebo', 'Depends on gazebo, turtlebot3-gazebo which is not available', '', d)}"
```

but because in our CI builds with ROS distro we don't include meta-qt5 (so rqt-drone-teleop is blacklisted by qt5 in ROS_WORLD_SKIP_GROUPS) and in webOS OSE builds we do include meta-qt5, but skip all pyqt5, so rqt-drone-teleop is still blacklisted there and doesn't trigger this error in any of our builds.

Will fix it. Feel free to report other cases like this or better send PRs fixing them.

Kirollos-Henry, 2021-09-02:
Sure I have much to do with ros so when I face anything new I will inform you.
this commit is not the full solution right?
because changing this line changes the error but didn't fix that error
I have got that:

ERROR: Nothing RPROVIDES 'rqt-drone-teleop' (but /home/Kirollos/proc-sdk/tisdk/sources/meta-ros/meta-ros1-noetic/recipes-core/packagegroups/packagegroup-ros-world-noetic.bb RDEPENDS on or otherwise requires it)
rqt-drone-teleop was skipped: Recipe is blacklisted: qt5: requires rqt-gui->qt-gui->qtbase which requires meta-qt5 to be included; pyqt5: rqt-gui-py->qt-gui->python-pyqt5 which requires pyqt5; gazebo: requires drone-wrapper which depends on gazebo

shr-project, 2021-09-02:
I was in a hurry and forgot to update the skip list in the packagegroups, see updated PR.

Kirollos-Henry, 2021-09-02:
Thank you so much I have just added it myself to the recipe, it is now building just fine if I found any error in the build I will report to you.

I have two small question as I notices that the full build is so big.

if I want to build minimum required to run ros and add packages after that is all I have to do is to add "ros-core" in my image recipe or how to add it in my image recipe as I didn't find any recipe inside noetic that I think is relevant to be minmum build?
and is there a way to add packages or just search for its recipe and add it to my image recipe in the image install too?

shr-project, 2021-09-02:
Adding ros-core package to your image should give you bare minimum to pass sanity test described here https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#image-sanity-testing

That's what's installed in ros-image-core here (shared for all ROS distros):
https://github.com/ros/meta-ros/blob/master/meta-ros-common/recipes-core/images/ros-image-core.bb

closed 2021-09-02 by shr-project