ros/meta-ros #926: Error bitbaking ros-core with Yocto Dunfell 3.1.11

suhrid-s, 2021-09-09:
**Describe the bug**

Packages installed but not shipped error when bitbaking ros-core using Yocto Dunfell (3.1.11), however, no error is faced when bitbaking ros-core using Yocto Hardknott (3.3.2)

**To Reproduce**

1) After bitbaking base image, clone meta-ros and checkout dunfell/milestones/16 tag.

2) List of ros related Yocto layers included:

  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2 \
  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros-common \
  /home1/user/rity-dunfell/src/poky/../meta-ros/meta-ros-backports-gatesgarth \
  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2-galactic \

3) Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "dev-platform"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.2"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa73-cortexa53"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"

**Error faced**

ERROR: rosidl-runtime-c-2.2.1-2-r0 do_configure: Execution of '/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/temp/run.do_configure.40429' failed with exit code 1:
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
-- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ament_cmake_ros: 0.9.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/share/ament_cmake_ros/cmake)
-- Found PythonInterp: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.8.11", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3
-- Found rcutils: 4.0.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake)
CMake Error at /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'rcutils' exports the library 'rcutils' which couldn't be found
Call Stack (most recent call first):
  /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/build/CMakeFiles/CMakeOutput.log".
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/temp/log.do_configure.40429
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 9.3.0
| -- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Found ament_cmake_ros: 0.9.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/share/ament_cmake_ros/cmake)
| -- Found PythonInterp: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.8.11", minimum required is "3")
| -- Using PYTHON_EXECUTABLE: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3
| -- Found rcutils: 4.0.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake)
| CMake Error at /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
|   Package 'rcutils' exports the library 'rcutils' which couldn't be found
[... 444 lines cut ...]
3) Add meta-ros layers as above to bblayers.conf
4) bitbake ros-core
5) Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "dev-platform"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.2"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa73-cortexa53"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"

6) Result:
Initialising tasks: 100% |###################################################################################################| Time: 0:00:11
Sstate summary: Wanted 2 Local 2 Network 0 Missed 0 Current 1519 (100% match, 100% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 4866 tasks of which 4866 didn't need to be rerun and all succeeded.

Please let me know what I could be doing incorrectly while bitbaking ros-core with Dunfell 3.1.11.
Thanks!

shr-project, 2021-09-09:
```
/home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2
/home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros-common
/home1/user/rity-dunfell/src/poky/../meta-ros/meta-ros-backports-gatesgarth
/home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2-galactic \
```

Why you're missing rity-v21.2 and rity-dunfell directories and what's the difference between them?

shr-project, 2021-09-09:
Also
```
BB_VERSION = "1.50.0"
```
dunfell is using 1.46 bitbake, so you're mixing wrong versions together, surprising that you get so far to actually start the build without parsing errors.

suhrid-s, 2021-09-09:
Hi,
The expected behavior portion is for my hardknott build. BB_VERSION = "1.50.0" is for the hardknott build which works. My error is regarding dunfell.
Thanks!

suhrid-s, 2021-09-09:
The meta-ros repo in both folders (rity-dunfell and rity-v21.2) are the same.
I replaced "rity-dunfell" with "rity-v21.2" and got the same errors.

closed 2021-09-09 by suhrid-s