ros/meta-ros #927: Error bitbaking ros-core with Yocto Dunfell 3.1.11

suhrid-s, 2021-09-09:
**Describe the bug**
Packages installed but not shipped error when bitbaking ros-core using Yocto Dunfell (3.1.11), however, no error is faced when bitbaking ros-core using Yocto Hardknott (3.3.2)

**To Reproduce**

1) After bitbaking base image, clone meta-ros and checkout dunfell/milestones/16 tag.

2) List of ros related Yocto layers included:

  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2 \
  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros-common \
  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros-backports-gatesgarth \
  /home1/user/rity-v21.2/src/poky/../meta-ros/meta-ros2-galactic \

3) Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "dev-platform"
DISTRO               = "rity-demo"
DISTRO_VERSION       = "21.3-dev"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
DISTRO_NAME          = "Rity Demo Layer"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"

**Error faced**

ERROR: rosidl-runtime-c-2.2.1-2-r0 do_configure: Execution of '/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/temp/run.do_configure.40429' failed with exit code 1:
-- The C compiler identification is GNU 9.3.0
-- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
-- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ament_cmake_ros: 0.9.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/share/ament_cmake_ros/cmake)
-- Found PythonInterp: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.8.11", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3
-- Found rcutils: 4.0.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake)
CMake Error at /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'rcutils' exports the library 'rcutils' which couldn't be found
Call Stack (most recent call first):
  /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/rcutilsConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/build/CMakeFiles/CMakeOutput.log".
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/temp/log.do_configure.40429
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 9.3.0
| -- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Found ament_cmake_ros: 0.9.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/share/ament_cmake_ros/cmake)
| -- Found PythonInterp: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found suitable version "3.8.11", minimum required is "3")
| -- Using PYTHON_EXECUTABLE: /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot-native/usr/bin/python3-native/python3
| -- Found rcutils: 4.0.2 (/home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake)
| CMake Error at /home1/user/rity-v21.2/nn/tmp/work/aarch64-poky-linux/rosidl-runtime-c/2.2.1-2-r0/recipe-sysroot/usr/share/rcutils/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
|   Package 'rcutils' exports the library 'rcutils' which couldn't be found
| Call Stack (most recent call first):
[... 440 lines cut ...]
For bitbaking ros-core using Hardknott image,
1) Bitbake base image
2) Clone meta-ros and checkout "hardknott" branch.
3) Add meta-ros layers as above to bblayers.conf
4) bitbake ros-core
5) Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "dev-platform"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.2"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa73-cortexa53"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"

6) Result:
Initialising tasks: 100% |###################################################################################################| Time: 0:00:11
Sstate summary: Wanted 2 Local 2 Network 0 Missed 0 Current 1519 (100% match, 100% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 4866 tasks of which 4866 didn't need to be rerun and all succeeded.

shr-project, 2021-09-10:
meta-ros-common/conf/layer.conf has ros-backports-hardknott-layer in LAYERDEPENDS_ros-common-layer but I don't see backports-hardknott in your list of layers, do you have some local modifications in meta-ros-common? or did you forgot to mention meta-ros-backports-hardknott layer?

shr-project, 2021-09-10:
meta-ros-backports-hardknott layer and its dependency was introduced after dunfell/milestones/16, so that explains why it was missing in your setup.

But I've tried to build qemux86-64 and qemuarm64 with default poky setup and dunfell/milestones/16 and rcpputils builds fine here, so it has to be something introduced by your poky modifications or other layers you've included.

```
poky$ bitbake -k rcpputils
Parsing recipes: 100% |#############################################################################################################################################################################################################################################################| Time: 0:00:09
Parsing of 2551 .bb files complete (0 cached, 2551 parsed). 3747 targets, 214 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "qemuarm64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.11"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
DISTRO_NAME          = "Poky (Yocto Project Reference Distro)"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta-ros2-galactic
meta-ros2
meta-ros-backports-gatesgarth
meta-ros-common      = "HEAD:67fd21d38192427a3183527220baaf024bb9aeae"
meta-oe
meta-python          = "dunfell:5c347d8ce425dcb4896e6f873810b8bfff5e4e92"
meta
meta-poky
meta-yocto-bsp       = "dunfell:60cfe38b513d474714500d1f5fcf408139c6a146"

Initialising tasks: 100% |##########################################################################################################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 40 Found 0 Missed 40 Current 180 (0% match, 81% complete)
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 938 tasks of which 836 didn't need to be rerun and all succeeded.
```

It can be many things including your dev-platform MACHINE or your rity DISTRO, I cannot guess what you're doing there.

I recommend to build with default Poky from dunfell (like you did for hardknott) and then compare the WORKDIR of some failing recipes to see what is different in your setup and go from there.

suhrid-s, 2021-09-10:
Hi,
Thanks a lot for your prompt reply!

Yes, I did try to build these recipes with default distro (Poky) and machine set to qemuarm64 and the build was successful.

On closer inspection, it seems like "multilib" gets enabled during rcutils bitbake. The files are written to /usr/lib64 instead of /usr/lib which causes other things to fail one after the other.
This does seem weird as it is only the ros recipes that seem to be affected by multilib. Is there something I should disable?

Thanks!

suhrid-s, 2021-09-14:
Hi,
I could bitbake ros-core.
The issue was indeed because of multilib being enabled. Some packages were being written to /usr/lib64 instead of /usr/lib because of which paths to the cmake files were incorrect.

Thanks for your help!

closed 2021-09-14 by suhrid-s

orascheg, 2022-06-09:
I also had this problem. A hint for all having this issue too:
Remove any line like:
MULTILIBS = "multilib:lib32"
from your configuration. I was able to build after that.