ros/meta-ros #938: Packaging geographiclib datasets

windelbouwman, 2021-09-30:
I just faced the issue where mavros requires the presence of some geographiclib datasets. Those can be installed with this script:

https://github.com/mavlink/mavros/blob/ros2/mavros/scripts/install_geographiclib_datasets.sh

I'm willing to supply a patch as a bbappend to mavros to include these files, but is this the best place?

shr-project, 2021-09-30:
From the script it looks like we would need native version of ./meta-ros-common/recipes-extended/geographiclib/geographiclib_1.48.bb

and then it might make sense to add some PN-dataset package into geographiclib recipe and install the datasets there as the script as-is won't work in cross build environment anyway.

Then mavros could just have runtime dependency on geographiclib-dataset.

windelbouwman, 2021-09-30:
Yes, I agree, but I checked the scripts `geographiclib-get-*` and these are simple wrappers around wget.

The way I solved it now, is adding a `geographiclib-data` recipe, and added a `RDEPENDS` on `mavros`.

This is my `geographiclib-data` recipe. It is highly customized to mavros, so I'm in doubt whether it should be part of mavros instead, maybe the recipe should be named `mavros-geographiclib-datasets`?
```
# Mavros selects these files:
# geoids egm96-5
# gravity egm96
# magnetic emm2015

DESCRIPTION = "Data files for geographiclib."
SECTION = "devel"

SRC_URI = " \
    https://downloads.sourceforge.net/project/geographiclib/geoids-distrib/egm96-5.tar.bz2?use_mirror=autoselect;name=geoids;downloadfilename=egm96-5.tar.bz2 \
    https://downloads.sourceforge.net/project/geographiclib/gravity-distrib/egm96.tar.bz2?use_mirror=autoselect;name=gravity;downloadfilename=egm96.tar.bz2 \
    https://downloads.sourceforge.net/project/geographiclib/magnetic-distrib/emm2015.tar.bz2?use_mirror=autoselect;name=magnetic;downloadfilename=emm2015.tar.bz2 \
"

SRC_URI[geoids.md5sum] = "6ae1c02c0506a686c9ad33be42cad0ae"
SRC_URI[geoids.sha256sum] = "c46224f8f723dc915d97179f4e1580a98d6c742fe2b82cd8fef0ecaaad13e614"
SRC_URI[gravity.md5sum] = "dcc1ab8e3433bd0add901b2f3e5caa76"
SRC_URI[gravity.sha256sum] = "6fea4c6bd56ff8ac53dbdad8d5dd505c855471d0354c4abc5c5fe048bf8350c1"
SRC_URI[magnetic.md5sum] = "18065cd3cc7369293d2a43c0c8f45315"
SRC_URI[magnetic.sha256sum] = "8e71a9704c5f2714bb65581df68e30f0d84d0ad17286d00efb782e7232334c3f"

do_install() {
    install -d -m 0755 ${D}/${datadir}/GeographicLib/geoids
    install -d -m 0755 ${D}/${datadir}/GeographicLib/gravity
    install -d -m 0755 ${D}/${datadir}/GeographicLib/magnetic
    install -D -m 0644 ${WORKDIR}/geoids/* ${D}/${datadir}/GeographicLib/geoids
    install -D -m 0644 ${WORKDIR}/gravity/* ${D}/${datadir}/GeographicLib/gravity
    install -D -m 0644 ${WORKDIR}/magnetic/* ${D}/${datadir}/GeographicLib/magnetic
}

FILES:${PN} = " \
    ${datadir}/GeographicLib \
"
```

shr-project, 2021-09-30:
> Yes, I agree, but I checked the scripts `geographiclib-get-*` and these are simple wrappers around wget.

Ah then OK, fetching with bitbake is much better, but please use some prefix like `mavros-geographiclib-datasets-` for the downloadfilename params. So that they are recognizable in downloads directory.

> The way I solved it now, is adding a `geographiclib-data` recipe, and added a `RDEPENDS` on `mavros`.

`RDEPENDS:${PN} += "mavros-geographiclib-datasets"` in `mavros` recipe, right? Not vice-versa.

> This is my `geographiclib-data` recipe. It is highly customized to mavros, so I'm in doubt whether it should be part of mavros instead, maybe the recipe should be named `mavros-geographiclib-datasets`?

As it's only fetching few files I don't mind either extra PACKAGE inside existing mavros recipe (through bbappend) or separate recipe in meta-ros-common where geographiclib recipe lives (if all mavros versions currently in meta-ros use the same - if there are significant differences, then bbappend tailored for each version would be probably easier to manage).

> DESCRIPTION = "Data files for geographiclib."

Use SUMMARY instead and mention that it's including only what mavros uses.

Please send PR we can continue the discussion there.

windelbouwman, 2021-12-21:
I consider this item done.

closed 2021-12-21 by windelbouwman