ros/meta-ros #946: Question: How to cross-compile custom ROS2 (foxy) packagre for Yocto Hardknott running on iMX8M-Mini SoC

bojankoce, 2021-10-13:
Hello, guys!

I have a headache finding out what would be the most efficient way to develop the ROS2 application that will execute on a target machine (DART-MX8M-Mini SoM) running Yocto Hardknott - [link](https://www.variscite.com/product/system-on-module-som/cortex-a53-krait/dart-mx8m-mini-nxp-i-mx8m-mini/?utm_source=google&utm_medium=cpc&utm_campaign=291971056&utm_content=532110069537&utm_term=&gclid=CjwKCAjwh5qLBhALEiwAioods_4YBjAwe9mqeq7pdZ58inI733nhLfbFsKz_EaNXLnY88O6qknOWsRoCuT0QAvD_BwE). 

My Host environment is Ubuntu 20.04. I can source from there the environment setup script provided by SoM manufacturer that will properly set the cross-toolchain (cross-compiler, sysroot...). More details [here](https://variwiki.com/index.php?title=Yocto_Toolchain_installation&release=RELEASE_HARDKNOTT_V1.0_DART-MX8M-MINI). 

Also, ros-meta layer is bitbaked into the Yocto image. 

Would it be possible for me to cross-compile the custom ROS2 package from my Host environment, generate binary executable and transfer it to the target machine running Yocto for testing/execution? If affirmative, what would be the steps to follow?

Thank you very much for your time and efforts. 
Looking forward to reading from you.

shr-project, 2021-10-13:
More typical use would be to create a recipe for your custom ROS2 package in your integration metadata layer and build it with bitbake as all other recipes in meta-ros are built.

For better development experience you can use `devtool` https://docs.yoctoproject.org/ref-manual/devtool-reference.html which simplifies building inside OE development as well as deploying the binaries you want to test to target machine (and remote debugging with a bit of work as well).

Using separate cross-toolchain SDK should be possible as well, but depends on how it's set by SoM and there would be a lot of work on meta-ros side as well (to provide nativesdk extends where needed as well as making sure the environment set by SDKs environment script is correct - this is very fragile in OE environment already with ament searching native and target sysroots in a bit mysterious ways sometimes or python modules from target sysroot being used instead of native ones during build - in separate SDK this might get even worse to fix properly - I hope to eventually clean this when migrating ROS2 to use ros_opt_prefix like ROS1 does).

amamory, 2022-03-16:
late reply, but for future references.

I am not an ROS/Yocto expert but I can share the path that work for me, without much effort. 

I tried to use meta-ros itself as a reference. So i looked for the simplest recipe i could find. Eventually i found this one
[https://github.com/ros/meta-ros/blob/zeus/meta-ros2-foxy/generated-recipes/demos/dummy-sensors_0.9.3-1.bb](https://github.com/ros/meta-ros/blob/zeus/meta-ros2-foxy/generated-recipes/demos/dummy-sensors_0.9.3-1.bb)
that worked fine for my 1st ROS2 Yocto recipe. After the 1st one works, you can keep doing a similar approach, which means finding a similar recipe to the one you want o create.

I hope this can help ....

SergeyFabric, 2022-09-12:
@shr-project Hi, do you know if there some news on SDK development? 
I working with couple of programmers at the same time so we need the SDK for a better workflow.

robwoolley, 2024-11-21:
@SergeyFabric Support for SDK development has been merged into the next branches for kirkstone, scarthgap, styhead, and walnascar (master).   Our testing has been primarily on master+rolling, but the changes exist for all combinations so it should work.

If you have any questions about it, happy to help answer them.

closed 2024-11-21 by robwoolley