ros/meta-ros #954: ros2-foxy component

yiyibst, 2021-10-20:
I want to use the basic communication and compilation features of ros2-foxy, what components do I need to add? And it is better to add components that do not depend on fastrtps, but only on cyclonedds.

amamory, 2022-03-16:
late answer, but just for documenting this answer for future references ...

This tutorial is probably not for the platform you use, but this bbappend file
https://www.hackster.io/jlamperez10/ros-2-in-kria-kv260-with-petalinux-2021-2-92ec3a#toc-3---create-a-custom-image-containing-ros-2-packages-3
contains a *minimal* working set of recipes for both fastrtps and cyclonedds. As you can imagine, the *absolute minimal* set of recipes will depend on your app requirements. But i am sure you can start w this set and testing the removal of some recipes.

robwoolley, 2023-11-17:
This question appears to have been resolved and is over 2 years old.

closed 2023-11-17 by robwoolley