ros/meta-ros #963: rmw_cyclonedds_cpp is not available at this   time

yiyibst, 2021-10-31:
I integrated ros2-foxy nativesdk-ament-cmake into my sdk, and integrated rmw-fastrtps-cpp, rmw-cyclonedds-cpp on the target and host side, but when I compile with aament, I get the following error:
Starting >>> cpp_pubsub
--- stderr: cpp_pubsub                         
CMake Error at /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rmw_implementation/cmake/rmw_implementation-extras.cmake:54 (message):
  The RMW implementation has been specified as 'rmw_cyclonedds_cpp' via
  environment variable 'RMW_IMPLEMENTATION', but it is not available at this
  time.

  

  Currently available middlewares:

  ''
Call Stack (most recent call first):
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rmw_implementation/cmake/rmw_implementationConfig.cmake:41 (include)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rcl/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rcl/cmake/rclConfig.cmake:41 (include)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/libstatistics_collector/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/libstatistics_collector/cmake/libstatistics_collectorConfig.cmake:41 (include)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rclcpp/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/dxy/dxy_virtualmachine/a1000_sdk/0905/sysroots/aarch64-bst-linux/usr/share/rclcpp/cmake/rclcppConfig.cmake:41 (include)
  CMakeLists.txt:14 (find_package)

---
Failed   <<< cpp_pubsub [1.86s, exited with code 1]

Why does this error occur and how do I test if fastrtps and cyclonedds are available at compile time?

Mustafa-AlHariry, 2022-07-26:
Hi yiyibst,
Could you tell me how you integrated "rmw-fastrtps-cpp" and "rmw-cyclonedds-cpp" ?

robwoolley, 2023-11-17:
Closing as this is over 2 years old without any updates.  Assuming the problem was resolved.

closed 2023-11-17 by robwoolley