ros/meta-ros #978: Build ROS2 package that contains custom interfaces (msg, srv)

bojankoce, 2021-11-23:
Hi, guys!

I have a [DART-MX8M-Mini machine](https://www.variscite.com/product/system-on-module-som/cortex-a53-krait/dart-mx8m-mini-nxp-i-mx8m-mini/?utm_source=google&utm_medium=cpc&utm_campaign=291971056&utm_content=532110069537&utm_term=&gclid=CjwKCAjwh5qLBhALEiwAioods_4YBjAwe9mqeq7pdZ58inI733nhLfbFsKz_EaNXLnY88O6qknOWsRoCuT0QAvD_BwE) running Yocto Hardknott. 

I properly added `meta-ros` layer and ROS2 foxy distro to our Yocto build by following instructions from [here ](https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project)([Sanity Tests](https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#ros-2-sanity-test) pass successfully).

With the help of `devtool`, I was able to build my custom ROS2 package (containing simple `talker` and `listener` nodes - [link](https://github.com/bojankoce/my_first_yocto_pkg)). For that to happen, I got some great support from the Yocto community - [link](https://lists.yoctoproject.org/g/yocto/topic/cross_compile_custom_ros2/86906676?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,86906676,previd=1636382356558417769,nextid=1636039783275782835&previd=1636382356558417769&nextid=1636039783275782835). 

Here is the content of `my-first-yocto-pkg_git.bb` recipe that enabled me to download my source code from the git and properly build the ROS2 package:
```
# Recipe created by recipetool
# This is the basis of a recipe and may need further editing in order to be fully functional.
# (Feel free to remove these comments when editing.)

# Unable to find any files that looked like license statements. Check the accompanying
# documentation and source headers and set LICENSE and LIC_FILES_CHKSUM accordingly.
#
# NOTE: LICENSE is being set to "CLOSED" to allow you to at least start building - if
# this is not accurate with respect to the licensing of the software being built (it
# will not be in most cases) you must specify the correct value before using this
# recipe for anything other than initial testing/development!

DESCRIPTION = "Examples of minimal publisher/subscriber using rclcpp."
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://package.xml;beginline=8;endline=8;md5=12c26a18c7f493fdc7e8a93b16b7c04f"

SRC_URI = "git://github.com/bojankoce/my_first_yocto_pkg;protocol=https"

# Modify these as desired
PV = "0.1.0"
SRCREV = "1312445de2e6861d9561c0f89f4827b94c2ff6b1"

S = "${WORKDIR}/git"

# NOTE: unable to map the following CMake package dependencies: rclcpp ament_lint_auto std_msgs ros_ament_cmake
inherit ros_distro_foxy
inherit ros_superflore_generated
inherit ros_ament_cmake

ROS_BUILD_DEPENDS = " \
    rclcpp \
    std-msgs \
"

ROS_BUILDTOOL_DEPENDS = " \
	ament-cmake-native \
"

ROS_EXPORT_DEPENDS = ""

ROS_BUILDTOOL_EXPORT_DEPENDS = ""

ROS_EXEC_DEPENDS = " \
    launch-ros \
    std-msgs \
"

DEPENDS = "${ROS_BUILD_DEPENDS} ${ROS_BUILDTOOL_DEPENDS}"

DEPENDS += "${ROS_EXPORT_DEPENDS} ${ROS_BUILDTOOL_EXPORT_DEPENDS}"

# Specify any options you want to pass to cmake using EXTRA_OECMAKE:
#EXTRA_OECMAKE = ""
```

I currently have a hard time building my ROS2 package that contains only custom ROS2 interfaces (messages and services). 
`CMakeLists.txt` file of that package contains the following lines that are causing the build errors:

```
find_package(rosidl_default_generators REQUIRED)
...

rosidl_generate_interfaces(${PROJECT_NAME}
  "msg/MessageX.msg"
  "msg/MessageY.msg"
  "srv/SrvX.srv"
  "srv/SrvY.srv"
 )
```

Do you have any idea about what should be included as `inherit xxxxx` or within `ROS_BUILD_DEPENDS`/`ROS_BUILDTOOL_DEPENDS `/`ROS_EXEC_DEPENDS` ... in order to successfully generate the ROS2 package containing custom interfaces?

Thanks in advance for your time and efforts. It is appreciated. 

Sincerely,
Bojan.

GeorgUr, 2022-04-06:
Hey! I am currently facing the same issue. Can you provide more information about the build error?

yimlaisum2014, 2024-01-10:
Hi, everyone. I have same issue too. Does anyone solve it?

bartoszptak, 2024-01-10:
> Hi, everyone. I have same issue too. Does anyone solve it?

Hi @yimlaisum2014,
I think the issue is still not solved. I workaround in that way [https://github.com/PUTvision/put-meta-husky/blob/master/husky-msgs_git.bb#L44C19-L44C33](https://github.com/PUTvision/put-meta-husky/blob/master/husky-msgs_git.bb#L44C19-L44C33). It is not a very practical solution, but I built successfully messages in yocto/petalinux.

robwoolley, 2024-01-16:
@yimlaisum2014 @bartoszptak Thanks for posting comments on this.  Sorry for the late response, I'm still getting used to the GitHub UI and didn't see this.

In the work I've been doing to update meta-ros, the *-msgs recipes have been the most trouble.  At first, I added the dependencies manually, but recently I have been adding rosidl-default-runtime or rosidl-default-generators to the ROS_BUILDTOOL_DEPENDS.   This has been supplying the dependencies it needs.  I have seen in newer ROS 2 distributions; they also require a dependency on service-msgs.

I suspect that the problem is related to how superflore generates the base recipes and changes in the default dependencies that are needed.   After I finish the work getting ros-image-world to work across all the supported combinations (which is soon), I plan to take a look at fixing superflore next.

ctgulzh, 2025-09-03:
Hi everyone, is there a definitive solution to this problem?