ros/meta-ros #989: populate_sdk does not generate include folder

mehmety888, 2022-01-06:
I have an existing OpenEmbedded Project, and an image called core-image-production.

I followed the instructions below, added layer meta-ros to conf/bblayers.conf and built the image.
https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions

I also added the following line to core-image-production.bb
require path/to/image/ros1-image-roslaunch.bb

It builds without problems. However, it does not generate include folder at opt/ros/melodic/

What should I do for it to generate an include folder ?

Thanks,

windelbouwman, 2022-01-10:
With a yocto SDK build, you do not get the `/opt/ros/<distro>` folder, the ros libraries and headers are stored in the sysroot folder for the target under `/usr/local/oecore-x86_64/sysroots/`.

mehmety888, 2022-01-10:
@windelbouwman I know, it should be at here: /usr/local/oecore-x86_64/sysroots/aarch64-poky-linux/opt/ros/melodic

  These are the folders inside melodic folder: bin, etc, lib, share, .catkin. There should also be an include folder but there isn't.

windelbouwman, 2022-01-10:
In my case I have to use this include path: `/usr/local/oecore-x86_64/sysroots/corei7-64-oe-linux/usr/include/`, which includes both linux generic headers, as well as the ros package headers. Btw, I'm using ROS2, maybe this is different with ROS1.

robwoolley, 2026-11-02:
Closing this issue as it seems to be resolved.  There is now a full ROS directory structure under /opt/ros/<DISTRO>

closed 2026-11-02 by robwoolley