ros/meta-ros #991: Cannot run ros-image-core

KansaiUser, 2022-01-25:
**Describe the bug**
Once ros-image-core is built according to your instructions, it fails to run

**To Reproduce**
I am building for MACHINE qemux86 (originally I made a mistake and built it for qemux86-64. After that I checked it is not supported by ros-image-core. Still the same error appeared)
I used
ros1-melodic-dunfell.mcf

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "i686-oe-linux"
MACHINE              = "qemux86"
DISTRO               = "ros1"
DISTRO_VERSION       = "1.melodic.2021-11-08.28-16+3.1.12-25.1-dunfell"
TUNE_FEATURES        = "m32 core2"
TARGET_FPU           = ""
DISTRO_NAME          = "Robot Operating System (ROS 1) with OpenEmbedded Linux"
ROS_DISTRO           = "melodic"
ROS_VERSION          = "1"
ROS_PYTHON_VERSION   = "2"
meta-raspberrypi     = "dunfell:934064a01903b2ba9a82be93b3f0efdb4543a0e8"
meta-ros1-melodic    
meta-ros1            
meta-ros-common      
meta-ros-backports-hardknott 
meta-ros-backports-gatesgarth 
meta-ros-python2     = "dunfell:ea3596dbc1eae450f49b1b0074e970f1ad2eb233"
meta-python2         = "dunfell:b901080cf57d9a7f5476ab4d96e56c30db8170a8"
meta-python          
meta-oe              = "dunfell:69f94af4d91215e7d4e225bab54bf3bcfee42f1c"
meta                 = "dunfell:f61fcb6e176f19f1e768ce63a693f238713c8887"

```

In your instructions, you did not specify how to run the image so I did

```
runqemu qemux86 ros-image-core
runqemu - INFO - Running MACHINE=qemux86 bitbake -e ...
runqemu - ERROR - Failed to find /opt/yocto/ros_workspace/tmp/deploy/images/qemux86/ros-image-core-qemux86.qemuboot.conf (wrong image name or BSP does not support running under qemu?).
runqemu - INFO - Cleaning up
```
If more info from a particular log is needed please specify.

The error clearly indicates it is looking for the file `opt/yocto/ros_workspace/tmp/deploy/images/qemux86/ros-image-core-qemux86.qemuboot.conf `
This file does not exist. Instead

```
ls tmp/deploy/images/qemux86/
bzImage                                                                          ros-image-core-melodic-qemux86-20220125001225.qemuboot.conf    ros-image-core-melodic-qemux86.ext4
bzImage--5.4.158+gitAUTOINC+db8bfc3a10_76404f1ae5-r0-qemux86-20220125001225.bin  ros-image-core-melodic-qemux86-20220125001225.rootfs.ext4      ros-image-core-melodic-qemux86.manifest
bzImage-qemux86.bin                                                              ros-image-core-melodic-qemux86-20220125001225.rootfs.manifest  ros-image-core-melodic-qemux86.qemuboot.conf
modules--5.4.158+gitAUTOINC+db8bfc3a10_76404f1ae5-r0-qemux86-20220125001225.tgz  ros-image-core-melodic-qemux86-20220125001225.rootfs.tar.bz2   ros-image-core-melodic-qemux86.tar.bz2
modules-qemux86.tgz                                                              ros-image-core-melodic-qemux86-20220125001225.testdata.json    ros-image-core-melodic-qemux86.testdata.json
/opt/yocto/ros_workspace[newDevelop]☢$ 
```
As you can see the file is `ros-image-core-melodic-qemux86-20220125001225.qemuboot.conf` or `ros-image-core-melodic-qemux86.qemuboot.conf`

**Expected behavior**
I expected the image to run like normal qemux86 yocto images do. I don't know (it is not clearly explained) how to proceed with ros after that but I will experiment

Please advice

KansaiUser, 2022-01-25:
Continuing investigating the problem, even though I built the image as

`bitbake -k  ros-image-core`

I found _strangely_ that I can run the image doing 

`runqemu -d qemux86 ros-image-core-melodic`

Doing that, the problem above does not happen but instead I got another problem

```
runqemu -d qemux86 ros-image-core-melodic
runqemu - DEBUG - Assuming MACHINE = qemux86
runqemu - INFO - Running MACHINE=qemux86 bitbake -e ...
runqemu - DEBUG - MACHINE: qemux86
runqemu - DEBUG - DEPLOY_DIR_IMAGE: /opt/yocto/ros_workspace2/tmp/deploy/images/qemux86
runqemu - DEBUG - CONFFILE: /opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.qemuboot.conf
runqemu - INFO - Overriding conf file setting of STAGING_DIR_NATIVE to /opt/yocto/ros_workspace2/tmp/work/core2-32-oe-linux/defaultpkgname/1.0-r0/recipe-sysroot-native from Bitbake environment
runqemu - INFO - Continuing with the following parameters:
KERNEL: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/bzImage--5.4.158+gitAUTOINC+db8bfc3a10_76404f1ae5-r0-qemux86-20220125085622.bin]
MACHINE: [qemux86]
FSTYPE: [ext4]
ROOTFS: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.ext4]
CONFFILE: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.qemuboot.conf]

runqemu - ERROR - runqemu-ifup: /opt/yocto/ros_workspace2/openembedded-core/scripts/runqemu-ifup
runqemu - ERROR - runqemu-ifdown: /opt/yocto/ros_workspace2/openembedded-core/scripts/runqemu-ifdown
runqemu - ERROR - ip: None
runqemu - ERROR - In order for this script to dynamically infer paths
 kernels or filesystem images, you either need bitbake in your PATH
 or to source oe-init-build-env before running this script.

 Dynamic path inference can be avoided by passing a *.qemuboot.conf to
 runqemu, i.e. `runqemu /path/to/my-image-name.qemuboot.conf`

 runqemu-ifup, runqemu-ifdown or ip not found
runqemu - INFO - Cleaning up
```
Apparently now, the problem is when setting the network tap??

KansaiUser, 2022-01-25:
I found that ip was not installed, so I did

`sudo apt update
sudo apt install iproute2`

with that, I could advance a little but get a new error

```
runqemu -d qemux86 ros-image-core-melodic
runqemu - DEBUG - Assuming MACHINE = qemux86
runqemu - INFO - Running MACHINE=qemux86 bitbake -e ...
runqemu - DEBUG - MACHINE: qemux86
runqemu - DEBUG - DEPLOY_DIR_IMAGE: /opt/yocto/ros_workspace2/tmp/deploy/images/qemux86
runqemu - DEBUG - CONFFILE: /opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.qemuboot.conf
runqemu - INFO - Overriding conf file setting of STAGING_DIR_NATIVE to /opt/yocto/ros_workspace2/tmp/work/core2-32-oe-linux/defaultpkgname/1.0-r0/recipe-sysroot-native from Bitbake environment
runqemu - INFO - Continuing with the following parameters:
KERNEL: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/bzImage--5.4.158+gitAUTOINC+db8bfc3a10_76404f1ae5-r0-qemux86-20220125085622.bin]
MACHINE: [qemux86]
FSTYPE: [ext4]
ROOTFS: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.ext4]
CONFFILE: [/opt/yocto/ros_workspace2/tmp/deploy/images/qemux86/ros-image-core-melodic-qemux86.qemuboot.conf]

runqemu - DEBUG - Running ('/sbin/ip', 'link')...
runqemu - INFO - Setting up tap interface under sudo
runqemu - ERROR - Setting up tap device failed:
Command '('sudo', '/opt/yocto/ros_workspace2/openembedded-core/scripts/runqemu-ifup', '1000', '7777', '/opt/yocto/ros_workspace2/tmp/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin')' returned non-zero exit status 1.
Run runqemu-gen-tapdevs to manually create one.
runqemu - INFO - Cleaning up
```
Again a problem when setting the network tap

robwoolley, 2023-11-17:
This problem seems to have been resolved.

closed 2023-11-17 by robwoolley