ros/meta-ros #996: libg2o without x11 and opengl

windelbouwman, 2022-04-08:
This is more of a brainstorm on how to use libg2o without x11 and opengl. In our system, we do not use the GUI and opengl, so I would like to be able to use libg2o without opengl and x11. The way this should probably work is by excluding x11 and opengl in the DISTRO_FEATURES variable. The problem is, that libg2o is in a blacklist, which requires these distro features.

We were able to work around this by adding a bbappend for libg2o like this:

```
DEPENDS_remove = "libglu mesa"
RDEPENDS:${PN}:remove = "libglu mesa"

EXTRA_OECMAKE += " -DG2O_USE_OPENGL=OFF"
```

Furthermore, we set the `PNBLACKLIST` for libg2o in our layer.conf:

```
PNBLACKLIST[libg2o] = ""
```

Ideally this should somehow be included in meta-ros, and libg2o should build with opengl support when the distro feature is set to support opengl. What would be a good way to do this, seeing that most recipes are generated, including the libg2o one?

shr-project, 2022-04-08:
It's possible to implement this in .bbappend which could be merged into meta-ros.

The ugly part is that the dependencies from package.xml cannot be conditional based on DISTRO_FEATURES, so the bbappend would need to use :remove like you did in your layer which is a bit ugly and I would recommend to at least use intermediate variable for removed deps (so that the :remove can be undone by setting this variable to empty).

And you can add PACKAGECONFIG[opengl] to handle EXTRA_OECMAKE and adding libglu+mesa, enable it based on DISTRO_FEATURES and then use contains conditional, something like:

PACKAGECONFIG ?= "${@bb.utils.filter('DISTRO_FEATURES', 'opengl', d)}"
PACKAGECONFIG[opengl] = "-DG2O_USE_OPENGL=ON,-DG2O_USE_OPENGL=OFF,mesa libglu"

DEPENDS_TO_REMOVE = "${@bb.utils.contains('PACKAGECONFIG', 'opengl', '', 'mesa libglu', d)}"
DEPENDS:remove = "${DEPENDS_TO_REMOVE}"
RDEPENDS:${PN}:remove = "${DEPENDS_TO_REMOVE}"

windelbouwman, 2022-05-02:
Thanks for the suggestions, I made a pull request here: https://github.com/ros/meta-ros/pull/1008

robwoolley, 2024-11-21:
The pull request has been merged into master-next.  Thanks!

closed 2024-11-21 by robwoolley