ros/meta-ros #997: ROS2 with RTI Connext DDS for ARMv8

abelal, 2022-04-13:
Hello community, we're looking to run Connext DDS on an ARM Cortex-A53 platform. There's a lot of confusion due to my lack knowledge of the whole ROS2/DDS integration so please bear with me.

I am able to build galactic (dunfell branch) of the repo following the instructions at https://github.com/ros/meta-ros/wiki/OpenEmbedded-Build-Instructions#add-meta-ros-to-an-existing-openembedded-project. I am also able to build my own images with the core packages i.e. ros-core. I see that for connext dds there's this RMW, rmw-connextdds, in the layer but unfortunately when I try to build it I get
```
bitbake rmw-connextdds
....
ERROR: Nothing PROVIDES 'rmw-connextdds'
rmw-connextdds was skipped: Recipe is blacklisted: connext: depends on rti-connext-dds which is not available

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
I tried to dig around but couldn't find much information on the subject.

So I believe this boils down to: how do I enable the RTI Connext DDS integration/support in ROS2 Yocto ARM Cortex-A53 build? Is it even possible with the dunfell/current metadata?

kuta42, 2022-04-29:
`connext` was first added to `ROS_WORLD_SKIP_GROUPS` in **foxy** by https://github.com/ros/meta-ros/commit/a33f7bea6930df32e663b8378f57b84f536f32b3 to blacklist a group of recipes (for **galactic**, they are: https://github.com/ros/meta-ros/blob/d775fdbadfecd9173485aa5bd13aff0b979aaf53/meta-ros2-galactic/recipes-core/packagegroups/packagegroup-ros-world-galactic.bb#L48-L59) that depend on `rti-connext-dds-5.3.1`, a recipe for which is not available in any upstream layer => you will have to create one yourself. Once you have created it, adding this to your `conf/local.conf`.
```
ROS_UNRESOLVED_DEP-rti-connext-dds-5.3.1 = "<NAME-OF-RECIPE>"
ROS_WORLD_SKIP_GROUPS:remove = "connext"
```
should allow you to build `rmw-connextdds`.

robwoolley, 2023-11-17:
This question appears to have been answered.

closed 2023-11-17 by robwoolley