ros/meta-ros #1000: Adding "colcon build" command to a Yocto image

BaptisteCNL, 2022-04-27:
**Describe the bug**
Unable to bitbake my image when including
TOOLCHAIN_HOST_TASK_append = " nativesdk-python3-colcon-common-extensions"
to my local.conf file.

**To Reproduce**
Using ROS2 foxy on Dunfell branch. I added meta-ros successfully to my image however, I'm unable to bitbake my image when adding colcon to the build process :

ERROR: Nothing RPROVIDES 'nativesdk-python3-coloredlogs' (but virtual:nativesdk:/home/chenal/Yocto-lab/meta-ros/meta-ros-common/recipes-devtools/colcon/python3-colcon-core_0.6.1.bb RDEPENDS on or otherwise requires it)
ERROR: Required build target 'mybot-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['mybot-image', 'buildtools-tarball', 'nativesdk-python3-colcon-common-extensions', 'nativesdk-python3-colcon-core', 'nativesdk-python3-coloredlogs']

**Expected behavior**
bitbake should successfully build "mybot-image"

windelbouwman, 2022-05-10:
There are two ways of using colcon:

1. Use colcon with a yocto SDK (preferred if you ask me)
2. Use colcon on the target device, and compile ros packages on the STM32MP1 core

In case of option 1, you would add the line you specified, and run `bitbake -c populate_sdk mybot-image`. The resulting SDK will end up in you deploy folder, and when installed and sourced should provide you with colcon on your host PC. This is not yet function, a pull request is waiting to be reviewed/merged, see for example this pull request: https://github.com/ros/meta-ros/pull/988

The problem you are facing specifically is that the `python3-coloredlogs` package does not have a BBCLASSEXTEND with nativesdk in it.

When choosing option 2, one should add the package `python3-colcon-common-extensions` to the `mybot-image` recipe.

BaptisteCNL, 2022-05-11:
Thanks a lot for your answer. 

I went for option 1 as you recommended (and also because it is inconveniant to build on the target device)
However when calling `bitbake -c populate_sdk mybot-image`, I've got an error which I can't seem to solve (I know it's not related to the original issue but I don't know how to solve this)

```
Transaction Summary
================================================================================
Install  2764 Packages

Total size: 2.5 G
Installed size: 15 G
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Error: Transaction check error:
  file /usr/include/lauxlib.h conflicts between attempted installs of lua-vendor-dev-0.0.1+2-r0.cortexa7t2hf_neon_vfpv4 and lua-dev-5.3.6-r0.cortexa7t2hf_neon_vfpv4
  file /usr/include/lua.h conflicts between attempted installs of lua-vendor-dev-0.0.1+2-r0.cortexa7t2hf_neon_vfpv4 and lua-dev-5.3.6-r0.cortexa7t2hf_neon_vfpv4
  file /usr/include/luaconf.h conflicts between attempted installs of lua-vendor-dev-0.0.1+2-r0.cortexa7t2hf_neon_vfpv4 and lua-dev-5.3.6-r0.cortexa7t2hf_neon_vfpv4
  file /usr/include/lualib.h conflicts between attempted installs of lua-vendor-dev-0.0.1+2-r0.cortexa7t2hf_neon_vfpv4 and lua-dev-5.3.6-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_chrono.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_filesystem.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_prg_exec_monitor.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_system.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_timer.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4
  file /usr/lib/libboost_unit_test_framework.so from install of boost-dev-1.72.0-r0.cortexa7t2hf_neon_vfpv4 conflicts with file from package microstrain-inertial-driver-2.0.6+1-r0.cortexa7t2hf_neon_vfpv4

Error Summary
-------------

```

I also tried option 2 and it worked for me (I removed` TOOLCHAIN_HOST_TASK_append = " nativesdk-python3-colcon-common-extensions"` and added `IMAGE_INSTALL_append = " cmake python3-colcon-common-extensions"` instead)
However when calling colcon build it gives me the following errors : `CMAKE_CXX_COMPILER not set, after EnableLanguage`
But I would rather go for option 1 if I can.

windelbouwman, 2022-05-11:
Thats an interesting error, it looks like the boost dynamic libraries are included in the microstrain-inertial-driver package, clearly, the so files of boost should be contained in the boost package, so something odd is happening in the microstrain-initial-driver package. You could check its work folder, and inspect the contents of the `image` subfolder to see what's contained in this package.

BaptisteCNL, 2022-05-13:
Thanks for your reply.
SDK is now fully generated. I built the environment by running the .sh script, then sourced the output files without error.
I have an other error which is :
```
File "/home/chenal/Yocto-lab/Wellbot_environment/sysroots/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/usr/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
```

I tried adding some modules to my image but without success, do you have a clue of what's missing ?

I added : 
```
TOOLCHAIN_TARGET_TASK_append = " \
    ament-cmake \
    ament-cmake-auto \
    ament-cmake-core \
    ament-cmake-export-definitions \
    ament-cmake-export-dependencies \
    ament-cmake-export-include-directories \
    ament-cmake-export-interfaces \
    ament-cmake-export-libraries \
    ament-cmake-export-link-flags \
    ament-cmake-export-targets \
    ament-cmake-gmock \
    ament-cmake-google-benchmark \
    ament-cmake-gtest \
    ament-cmake-include-directories \
    ament-cmake-libraries \
    ament-cmake-nose \
    ament-cmake-pytest \
    ament-cmake-python \
    ament-cmake-ros \
    ament-cmake-target-dependencies \
    ament-cmake-test \
    ament-cmake-version \
    ament-lint-auto \
    foonathan-memory-staticdev \
"
```

Once again, thanks a lot for your help and your work.

BaptisteCNL, 2022-05-17:
Adding

BaptisteCNL, 2022-05-17:
I'm using Yocto to compile my nodes. I took examples of the files present in meta ROS and managed to compile an image with my packages.
I added the package bb recipe and bb append files to the meta-ros folder where other generated recipes are, and I added my package name to the `superflore-ros-distro.inc` file. I also tried adding `IMAGE_INSTALL_append = " my-ros-package"` to my image recipe.

When using the generated image on my STM32MP157C-DK2, I see some files named "my-ros-package" in the `/usr/share/ament_index/resource_index/` folder but the folder `/usr/share/my-ros-package/` isn't in my filesystem.
(all meta-ros ros packages and nodes are in the /usr/share/ folder with their relatives files eg : launch.py, install.sh, ...)
Do you have an idea of what am I missing for my-ros-package to be usable in my image ?

Once again thank you for your time and your help.

BaptisteCNL, 2022-05-23:
What finally did the trick for me was adding thoses lines to my bb append file (instead of the superflore-ros-distro.inc file) :

```
ROS_SUPERFLORE_GENERATED_RECIPES += " imu-sensor "
ROS_SUPERFLORE_GENERATED_RECIPE_BASENAMES_WITH_COMPONENT += " imu-sensor/imu-sensor_1 "
ROS_SUPERFLORE_GENERATED_WORLD_PACKAGES = " imu-sensor "
ROS_SUPERFLORE_GENERATED_RECIPES_FOR_COMPONENTS = " imu-sensor "
```

jsb9945, 2023-12-02:
Hello, can may I ask you a question?

I entered the code below in the local.conf file and built the image.

'''
TOOLCHAIN_TARGET_TASK_append = " \
    ament-cmake \
"
'''

However, the following error occurred. How can i solve this?

(wrlinuxsdk-x86_64-aarch64) wkdtjdqls@wkdtjdqls-MS-7C94:~/ros2_ws$ colcon build                                                                                                              Starting >>> cpp_pubsub                                                                                                                                                                      --- stderr: cpp_pubsub                                                                                                                                                                       Traceback (most recent call last):                                                                                                                                                             File "/opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>                                 from ament_package.templates import get_environment_hook_template_path                                                                                                                   ModuleNotFoundError: No module named 'ament_package'                                                                                                                                         CMake Error at /opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):                                                                                                                                                                                                                          execute_process(/opt/windriver/wrlinux/21.20/sysroots/x86_64-wrlinuxsdk-linux/usr/bin/python3                                                                                                /opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py                                                            /home/wkdtjdqls/ros2_ws/build/cpp_pubsub/ament_cmake_package_templates/templates.cmake)                                                                                                      returned error code 1                                                                                                                                                                      Call Stack (most recent call first):                                                                                                                                                           /opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:41 (include)                                                         /opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)                                         /opt/windriver/wrlinux/21.20/sysroots/cortexa53-wrs-linux/usr/share/ament_cmake/cmake/ament_cmakeConfig.cmake:41 (include)                                                                   CMakeLists.txt:13 (find_package)                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                     ---                                                                                                                                                                                          Failed   <<< cpp_pubsub [0.07s, exited with code 1]                                                                                                                                                                                                                                                                                                                                       Summary: 0 packages finished [0.15s]                                                                                                                                                           1 package failed: cpp_pubsub                                                                                                                                                                 1 package had stderr output: cpp_pubsub           

Thank you.

OznOg, 2024-11-22:
> There are two ways of using colcon:
> 
>     Use colcon with a yocto SDK (preferred if you ask me)
>     Use colcon on the target device, and compile ros packages on the STM32MP1 core

Is there a reason why it would not be possible to use colcon in the do_compile() of a recipe?
When trying to simply build a project to be embedded, having to use the sdk or on the target is not really friendly.

robwoolley, 2024-11-23:
@OznOg I don’t think anything stops you from using colcon inside a recipe.  I think  it may negate the benefits from using bitbake. We currently use ament and CMake when executing the recipes.  What benefit would using colcon bring?

windelbouwman, 2024-11-23:
Agreed, I think you should use colcon or yocto as a top level build tool, not both. Running colcon inside a yocto recipe would probably cause colcon to build a whole lot of dependencies within a yocto recipe, so builds might be long, and artifacts my be cluttered or even corrupted with libraries build by colcon.

robwoolley, 2024-11-23:
Sounds like a common problem that many ROS developers would want to address.

Can you provide instructions or a reference example so I can recreate what you are doing today?

Thanks,
Rob

OznOg, 2024-11-25:
a very minimalistic example could be clone from `git@github.com:OznOg/example_meta_ros.git`

A git project mono  with "all" the ROS projects in it (here they were added as git subtree)
```
example_meta_ros
├── catkin
└── message_generation
```

then in my work directory, I clone meta project, and I simply run `colcon build` (I ignore build and install options for this example). This eventually builds everything following the dependencies, and install all in `install` subdirectory.

This works well on fedora, but is hardly achievable on yocto for now.

I hope you it helps,
Thank you for your help
Sebastien