ros/meta-ros #999: Make meta-ros layer compatible with kirkstone yocto version

freallearn, 2022-04-27:


shr-project, 2022-04-27:
I no longer maintain this layer, but @moto-timo has few changes to make this compatible https://github.com/moto-timo/meta-ros/commits/kirkstone

Future maintenance of meta-ros was discussed in OSRF TSC https://discourse.ros.org/t/ros-2-tsc-meeting-january-20th-2022/23986/3 , but I haven't seen any update since then (I don't follow discourse very closely).

shr-project, 2022-05-03:
https://github.com/ros/meta-ros/pull/1009

bchoineubility, 2023-07-14:
Hi,

I nearly can build ros-image-core with moto-timo/meta-ros  but, ros-image-world.

Maybe, it would be not tried in kirkstone and yocto yet.

please, if there are any update for it.

BR,
Mark

robwoolley, 2023-11-13:
I just pushed support for Noetic, Humble, and Iron to the kirkstone-next branch.   It builds with ros-image-core and getting ros-image-world working is desirable.   Any feedback you have would be valuable.  I encourage you to file issues with anything you find.

Regards,
Rob

bchoineubility, 2023-11-15:
@robwoolley Many thanks for this release. 

May I use kirkstone branch instead of kirkstone-next ?

BR,
Mark

robwoolley, 2023-11-15:
Yes, you can.  Both branches match at the moment.

I also just created a PR for Humble-Kirkstone that fixes the ROS packages to fix build failures when building for 64-bit machines: https://github.com/ros/meta-ros/pull/1057

I intend to merge it once I do more testing to make sure I haven't broken anything.