ros/meta-ros #1012: add meta-ros to existing Yocto project STM32MP1

SergeyFabric, 2022-05-14:
Hi, I try to add meta-ros to an Existing OpenEmbedded Project (STM32MP1) by the wiki page.
first of all, I don't understand step 2 "Add meta-ros[12]-<ROS_DISTRO> to your conf/bblayers.conf." of "Add meta-ros to an Existing OpenEmbedded Project " What does the "12" stand for?

then I just added this layers in different constellations  : 
bitbake-layers add-layer ../layers/meta-ros/meta-ros2-foxy/ 
bitbake-layers add-layer ../layers/meta-ros/meta-ros-common/

and i got the same output:
unparsed line: 'ROS_SUPERFLORE_GENERATED_BUILDTOOLS:remove = "ament-cmake-google-benchmark-native"'

**To Reproduce**

layers without meta-ros:
NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta-st-stm32mp-addons  /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-fabric/meta-st-stm32mp-addons  7
meta-oe               /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-oe  6
meta-initramfs        /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-initramfs  8
meta-multimedia       /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-multimedia  6
meta-networking       /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-networking  5
meta-webserver        /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-webserver  6
meta-filesystems      /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-filesystems  6
meta-perl             /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-perl  6
meta-python           /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-openembedded/meta-python  7
meta-st-stm32mp       /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-st-stm32mp  6
meta-qt5              /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-qt5  7
meta-st-openstlinux-cv  /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-fabric/meta-st-openstlinux-cv  6
meta-st-openstlinux   /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-st-openstlinux  5
meta-mt9v034          /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-fabric/meta-mt9v034  7
meta-wl18xx-layer     /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-fabric/meta-wl18xx-layer  7
meta-csr              /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-st/meta-fabric/meta-csr  6
meta                  /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/openembedded-core/meta  5

when i add meta-ros , meta-ros2-foxy to bblayers the output are:
NOTE: Starting bitbake server...
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/build-openstlinuxcv-stm32mp1-ground4-mx/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 27353 at 2022-05-14 19:18:16.683520 ---
ERROR: ParseError at /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-ros/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro.inc:43: unparsed line: 'ROS_SUPERFLORE_GENERATED_BUILDTOOLS:remove = "ament-cmake-google-benchmark-native"'
ERROR: ParseError at /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-ros/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro.inc:43: unparsed line: 'ROS_SUPERFLORE_GENERATED_BUILDTOOLS:remove = "ament-cmake-google-benchmark-native"'
ERROR: ParseError at /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-ros/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro.inc:43: unparsed line: 'ROS_SUPERFLORE_GENERATED_BUILDTOOLS:remove = "ament-cmake-google-benchmark-native"'

or when I run the same error : 
bitbake -p ros-core
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/build-openstlinuxcv-stm32mp1-ground4-mx/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 27408 at 2022-05-14 19:20:02.620360 ---
ERROR: ParseError at /home/sergey/yoctoWorkspace/openstlinux-5.4-dunfell-mp1-20-11-12/layers/meta-ros/meta-ros2-foxy/conf/ros-distro/include/foxy/ros-distro.inc:43: unparsed line: 'ROS_SUPERFLORE_GENERATED_BUILDTOOLS:remove = "ament-cmake-google-benchmark-native"'

why this meta so difficult to add ? it should be simple task like add to local.conf only the ros2-foxy or something like this at the end its only SW ...

shr-project, 2022-05-14:
your bitbake is too old, read:
https://github.com/ros/meta-ros/pull/902

SergeyFabric, 2022-05-14:
So if I update my STLinux to the last version that ST has it will help?
:openstlinux-5.10-dunfell-mp1-21-11-17 
Build process
OpenEmbedded LTS Dunfell (v3.1.11) - NEW

erikboto, 2022-05-18:
Yeah dunfell should be new enough, at least I use meta-ros with dunfell. Just make sure to use the dunfell branch of meta-ros.

closed 2022-09-01 by SergeyFabric