ros/meta-ros #1017: honister-humble build control-msgs failed

panxxhub, 2022-05-27:
**Describe the bug**
ERROR: control-msgs-3.0.0-1-r0 do_package: QA Issue: control-msgs: Files/directories were installed but not shipped in any package:

**To Reproduce**

**Expected behavior**

closed 2022-06-09 by panxxhub

vectorcrumb, 2023-04-17:
Were you able to resolve this issue? I'm also getting an error on control-msgs when trying to build a Petalinux image for the KR260 platform.

grsandeep85, 2023-04-17:
Hi @vectorcrumb,

We have fixed some ros2-control packages build issues last week, we will push the changes asap.

grsandeep85, 2023-04-20:
Hi @vectorcrumb,

Patches are pushed in repo  https://github.com/Xilinx/meta-ros/compare/b170fbcd5d85...bb62e31839b3

vectorcrumb, 2023-04-25:
Thanks @grsandeep85 !
Was is the reason behind using `inherit distutils3-base` and having to specify `python3-numpy-native` as a build dependency so often?

ctgulzh, 2025-09-01:
What are the specific steps for cross-compiling ROS 2 message and function packages on the host using the petalinux tool? Can they be compiled as libraries or apps? (See https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842475/PetaLinux+Yocto+Tips#PetaLinuxYoctoTips-HowtoAddPre-builtLibrariesinPetaLinuxorYoctoProjects.) Or do I need to use the devtool tool for cross-compilation?