ros/meta-ros #1018: Honister Humble Build fails on graphviz with demo image

coalman321, 2022-06-06:
**Describe the bug**
Configuration of graphviz fails during build of the recipe outlined in this article: https://news.accelerationrobotics.com/ros-2-humble-in-nvidia-xavier-agx-with-yocto/

**To Reproduce**

Branch: honister-humble
Target: Jetson Xavier AGX devkit
BSP: meta-tegra

Platform information: 
`coalman321@ros-ws:~/$ uname -a
Linux ros-ws 5.13.0-44-generic #49~20.04.1-Ubuntu SMP Wed May 18 18:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
`
Build Configuration:
```
BB_VERSION           = "1.52.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-oe4t-linux"
MACHINE              = "jetson-agx-xavier-devkit"
DISTRO               = "tegrademo"
DISTRO_VERSION       = "3.4.3+snapshot-20220606"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
DISTRO_NAME          = "OE4Tegra Demonstration Distro"
ROS_DISTRO           = "humble"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"
meta                 = "HEAD:580532cfd0c23e27a366c074e806fc0b6908b874"
meta-tegra           
contrib              = "HEAD:b2481c12ae0468c85b76332faa2922682245e185"
meta-oe              
meta-python          
meta-networking      
meta-filesystems     = "HEAD:a19d1802b15fe07e9b1e7584ff9af7c33bfe642a"
meta-virtualization  = "HEAD:bd7511c53b921c9ce4ba2fdb42778ca194ebc3e8"
meta-tegra-community = "HEAD:4a47d3798404904eec5299306d063a6ec753b158"
meta-tegra-support   
meta-demo-ci         
meta-tegrademo       = "honister:a70a14f6403955d62ca53ba19254efa75d29e6f6"
meta-ros2            
meta-ros2-humble     
meta-ros-common      = "honister-humble:20927e8fcf5cea099386a0933f95bb312b4f295d"
```

The specific error given by bitbake on exit is:
```
ERROR: graphviz-native-2.44.1-r0 do_configure: configure failed
ERROR: graphviz-native-2.44.1-r0 do_configure: ExecutionError('/home/coalman321/tegra-bsp-honister/build/tmp/work/x86_64-linux/graphviz-native/2.44.1-r0/temp/run.do_configure.1465090', 1, None, None)
ERROR: Logfile of failure stored in: /home/coalman321/tegra-bsp-honister/build/tmp/work/x86_64-linux/graphviz-native/2.44.1-r0/temp/log.do_configure.1465090
```

The full logs are located in this gist: https://gist.github.com/coalman321/d0295d4572815c4ab32dffe586d5c32b

**Expected behavior**
Expected to build a demo image as demonstrated in the article. 

Any help on this would be appreciated. Thank you.

@vmayoral Any suggestions here? I can build the tegrademo fine but one of the meta-ros dependencies seems to cause the issue above. I'm wondering if I am missing specific dependencies on the system to make this work correctly.

vmayoral, 2022-06-06:
Thanks for the report @coalman321, this is a nicely explained/written support ticket.

I've experienced this before myself and wrote about it at https://news.accelerationrobotics.com/ros2-humble-yocto-petalinux/#graphviz-dpkg-architecture-command-not-found.

In a nutshell, what worked for me:

```bash
sudo apt-get remove libtcl8.6
sudo apt-get update
sudo apt-get install blt libopencv-contrib-dev libopencv-dev libopencv-viz-dev libopencv-viz4.5d libtcl8.6
```

And then re-build again the `graphviz` recipe.

vmayoral, 2022-06-06:
@coalman321 if that addresses your issue and you move beyond this, can you comment in here if you successfully manage to build all the recipes for `humble`?

coalman321, 2022-06-06:
@vmayoral 
Yes that package change does fix the graphviz configuration for ubuntu 20.04. As a quick note, the libopencv-viz4.5d package does not exist in 20.04, and may be causing other issues as opencv from the meta-oe layer is now failing to compile.

I have put together another gist of the relevant logs here:  https://gist.github.com/coalman321/47e4a4eeb020f508922d47a011c8e4e1

The opencv error as far as I can tell looks to be an assembler error. I'm not sure why that would happen but it sure does cause a lot of terminal spam... Also there are failures on the example packages. I haven't taken a look at those failures yet though.

Build results.
```
NOTE: Tasks Summary: Attempted 10119 tasks of which 5888 didn't need to be rerun and 5 failed.

Summary: 5 tasks failed:
  /home/coalman321/tegra-bsp-honister/layers/meta-ros/meta-ros2-humble/generated-recipes/examples/examples-rclcpp-multithreaded-executor_0.15.0-2.bb:do_compile
  /home/coalman321/tegra-bsp-honister/layers/meta-ros/meta-ros2-humble/generated-recipes/demos/demo-nodes-cpp-rosnative_0.20.1-2.bb:do_compile
  /home/coalman321/tegra-bsp-honister/layers/meta-ros/meta-ros2-humble/generated-recipes/examples/examples-rclcpp-minimal-service_0.15.0-2.bb:do_compile
  /home/coalman321/tegra-bsp-honister/layers/meta-ros/meta-ros2-humble/generated-recipes/demos/demo-nodes-cpp_0.20.1-2.bb:do_compile
  /home/coalman321/tegra-bsp-honister/layers/meta-oe/recipes-support/opencv/opencv_4.5.2.bb:do_compile
Summary: There were 5 ERROR messages shown, returning a non-zero exit code.

```

vmayoral, 2022-06-07:
@coalman321, the details above don't help assessing your error. Can you paste your actual error while building the `opencv` recipe? Note that you need to make sure you have enough memory/SWAP to build certain recipes.

coalman321, 2022-06-07:
@vmayoral,
My apologies. I missed checking the log to see if it had relevant information. I located the cmake build logs for opencv. It looks like the assembler error may have been a distraction, as the errors that cmake shows look to be related to opencl support. The gist with actual logs is here: https://gist.github.com/coalman321/3afaaaca614acbd0aa97bbe7df981a09 . I would suggest looking at the CMake logs first, but the Ninja outputs have been provided as well.

vmayoral, 2022-06-07:
> The gist with actual logs is here: https://gist.github.com/coalman321/3afaaaca614acbd0aa97bbe7df981a09

Had a quick look. It's still a bit unclear but from the error dumps I couldn't see any issues related to `meta-ros` recipes. It appears you've got indeed some build issues with `opencv` (which is totally unrelated to `meta-ros`), but that's something else. Can't put time into debugging that, sorry.

It seems your current `meta-ros` issue has been addressed with the Humble branch @coalman321- If so, can we close this ticket? (feel free to open another one if other issues appear).

*I'd suggest bringing the opencv issue you're experiencing elsewhere (e.g. maybe to the https://github.com/OE4T/meta-tegra repo? you'll need to ask there)*.

coalman321, 2022-06-07:
I did some additional testing by re-running the build. It never ran out of memory during the compile and is currently only failing at the minimal service example now. I’m not sure if it was able to re-run the others to know if those successfully completed.

coalman321, 2022-06-07:
I think the original error here has been resolved on my end with the package change. It looks like the OpenCV failure was unrelated, and now gone so I’ll close the ticket out.