ros/meta-ros #1020: ROS2 Foxy build on Gatesgarth fails due to wrong layer name

orascheg, 2022-06-08:
At building ROS2 foxy with milestone 17 of the Gatesgarth branch the following error message is shown:

ERROR: The following layer directories do not exist:
ERROR:    /var/yocto/dev/meta-ros/meta-ros-backports-hardknott-layer
ERROR: Please check BBLAYERS in /var/yocto/dev/build/conf/bblayers.conf

Fast solution was to rename the directory in meta-ros
/var/yocto/dev/meta-ros/meta-ros-backports-hardknott
to
/var/yocto/dev/meta-ros/meta-ros-backports-hardknott-layer

Unfortunately I am not in the solution too deep and do not know where this typo is to be changed. My guess is that the meta-ros-backports-hardknott-layer/conf/layer.conf has to be adapted

shr-project, 2022-06-08:
How did you create your bblayers.conf? You should fix it there, not by renaming the layer.

orascheg, 2022-06-08:
Thanks for the extremely swift answer. Sorry, that I have not followed the reporting rules, I am new to this. The distro is built with kas and thus the bblayers.conf is generated:

kas build --target swu-devros2-update meta-devros2/conf/kas/base/devros2-base-settings.yml
2022-06-08 11:22:22 - INFO     - kas 3.0.2 started
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-devros2/conf/kas/base$ git rev-parse --show-toplevel
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-devros2/conf/kas/base$ git rev-parse --show-toplevel
2022-06-08 11:22:22 - WARNING  - Falling back to file-relative addressing of local include "../common-settings.yml"
2022-06-08 11:22:22 - WARNING  - Update your layer to repo-relative addressing to avoid this warning
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-devros2$ git rev-parse --show-toplevel
2022-06-08 11:22:22 - INFO     - Using /var/yocto/dev/meta-devros2 as root for repository meta-devros2
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-dev-common$ git remote set-url origin ssh://devops.dev.com:22/009/FirmwareDeviceBuilder/_git/meta-dev-common
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-base$ git remote set-url origin ssh://devops.dev.com:22/009/FirmwareDeviceBuilder/_git/meta-base
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-swupdate-dev$ git remote set-url origin ssh://devops.dev.com:22/009/FirmwareDeviceBuilder/_git/meta-swupdate-dev
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/poky$ git remote set-url origin git://git.yoctoproject.org/poky
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-openembedded$ git remote set-url origin https://git.openembedded.org/meta-openembedded
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-toradex-nxp$ git remote set-url origin https://git.toradex.com/cgit/meta-toradex-nxp.git/
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-toradex-bsp-common$ git remote set-url origin http://git.toradex.com/meta-toradex-bsp-common.git
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-freescale$ git remote set-url origin https://git.yoctoproject.org/git/meta-freescale
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-freescale-3rdparty$ git remote set-url origin https://github.com/Freescale/meta-freescale-3rdparty
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-swupdate$ git remote set-url origin https://github.com/sbabic/meta-swupdate/
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-ros$ git remote set-url origin https://github.com/ros/meta-ros.git
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-dev-common$ git cat-file -t main
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-base$ git cat-file -t main
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-swupdate-dev$ git cat-file -t main
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/poky$ git cat-file -t bc71ec0f1d97c983cb0a089b15e283c56de30844
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-openembedded$ git cat-file -t f3f7a5f1a4713f145107bb043e0d14cb3a51c62f
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-toradex-nxp$ git cat-file -t d26fac8da30aa1e530f11178f85ea285d329d442
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-toradex-bsp-common$ git cat-file -t a3debcd429139019c528932b0436854349b4fb86
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-freescale$ git cat-file -t d5a82541a97dd3a6b9116a1cd72f9965510a3de9
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-freescale-3rdparty$ git cat-file -t 4297672fe35c2c24350f0eaba3d909f5a2dcee4c
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-swupdate$ git cat-file -t 9b864daefacbebe6fb7dcf7373a3421b56b42994
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-ros$ git cat-file -t gatesgarth/milestones/17
2022-06-08 11:22:22 - INFO     - Repository meta-dev-common already contains main as commit
2022-06-08 11:22:22 - INFO     - Repository meta-base already contains main as commit
2022-06-08 11:22:22 - INFO     - Repository meta-swupdate-dev already contains main as commit
2022-06-08 11:22:22 - INFO     - Repository poky already contains bc71ec0f1d97c983cb0a089b15e283c56de30844 as commit
2022-06-08 11:22:22 - INFO     - Repository meta-openembedded already contains f3f7a5f1a4713f145107bb043e0d14cb3a51c62f as commit
2022-06-08 11:22:22 - INFO     - Repository meta-toradex-nxp already contains d26fac8da30aa1e530f11178f85ea285d329d442 as commit
2022-06-08 11:22:22 - INFO     - Repository meta-toradex-bsp-common already contains a3debcd429139019c528932b0436854349b4fb86 as commit
2022-06-08 11:22:22 - INFO     - Repository meta-freescale already contains d5a82541a97dd3a6b9116a1cd72f9965510a3de9 as commit
2022-06-08 11:22:22 - INFO     - Repository meta-freescale-3rdparty already contains 4297672fe35c2c24350f0eaba3d909f5a2dcee4c as commit
2022-06-08 11:22:22 - INFO     - Repository meta-swupdate already contains 9b864daefacbebe6fb7dcf7373a3421b56b42994 as commit
2022-06-08 11:22:22 - INFO     - Repository meta-ros already contains gatesgarth/milestones/17 as tag
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-devros2$ git rev-parse --show-toplevel
2022-06-08 11:22:22 - INFO     - Using /var/yocto/dev/meta-devros2 as root for repository meta-devros2
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-dev-common$ git status -s
2022-06-08 11:22:22 - INFO     - /var/yocto/dev/meta-dev-common$ git rev-parse --verify -q origin/main
2022-06-08 11:22:22 - INFO     - ac1ca051337a641df0c5f4fe1a113a2c1a49e466
[... cut ...]
2022-06-08 11:22:23 - INFO     - /var/yocto/dev/build$ /var/yocto/dev/poky/bitbake/bin/bitbake -c build swu-devros2-update
ERROR: The following layer directories do not exist:
ERROR:    /var/yocto/dev/meta-ros/meta-ros-backports-hardknott-layer
ERROR: Please check BBLAYERS in /var/yocto/dev/build/conf/bblayers.conf
2022-06-08 11:22:24 - ERROR    - Command returned non-zero exit status 1

I understand that I have got to find out why kas is generating a different name, sorry to bother you with it.

shr-project, 2022-06-08:
I don't know what meta-devros2 is, but I guess the issue is in devros2-base-settings.yml.

orascheg, 2022-06-08:
Thanks a lot for that hint. I have checked all layer.conf files, which are basically the same thus this was not causing the issue
The yml file defines the contents which are then generated into the bblayers.conf. I have errounously added the layer postfix:
  meta-ros:
    url: https://github.com/ros/meta-ros.git
    refspec: gatesgarth/milestones/17
    layers:
      meta-ros2:
      meta-ros2-foxy:
      meta-ros-common:
      meta-ros-backports-hardknott-layer:

Next time I will look even once more and follow the reporting rules.

I'd like to state that it is very appreciated that you maintain meta-ros!

closed 2022-06-08 by orascheg