ros/meta-ros #1023: ROS2 foxy on Yocto 3 (zeus)

folibis, 2022-07-26:
I have an image built using Yocto. The version is 3.0 (zeus) and unfortunately I can't change/upgrade it since that come from the board supplier. So I've cloned the zeus branch:
`git clone --single-branch --branch zeus https://github.com/ros/meta-ros.git`
and added the layer to bblayrs.conf:
` BBLAYERS += "${BSPDIR}/sources/meta-ros/meta-ros2-foxy"`
Unfortunately the build failed with error:
> ERROR: Layer ros2-foxy-layer is not compatible with the core layer which only supports these series: zeus

What did I wrong and how to fix that?

apbr, 2022-08-03:
Tried also adding `meta-ros-common` and `meta-ros2`?

DasRoteSkelett, 2022-08-23:
From reading through it:
```
meta-ros2-foxy/conf/layer.conf:LAYERSERIES_COMPAT_ros2-foxy-layer = "${ROS_OE_RELEASE_SERIES}"
```
and 
```
meta-ros-common/conf/ros-distro/ros-distro.conf:ROS_OE_RELEASE_SERIES = "zeus"
```
Which is required when including meta-ros-common. Hence, @apbr should be right, this should be fixed by including the other layers as well:
```
LAYERDEPENDS_ros2-foxy-layer = " \
    core \
    meta-python \
    openembedded-layer \
    ros-common-layer \
    ros2-layer \
"
```

robwoolley, 2023-11-17:
This appears to have been answered.  Also, foxy and zeus are no longer supported.

closed 2023-11-17 by robwoolley