ros/meta-ros #1030: meta-ros2-galactic  (foonathan-memory) build failure on Yocto dunfell for MACHINE POWER PC

sweepingman, 2022-11-07:
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "powerpc-fsl-linux"
MACHINE              = "p2020rdb"
DISTRO               = "fsl-qoriq"
DISTRO_VERSION       = "3.1.20"
TUNE_FEATURES        = "m32 ppce500v2 bigendian"
TARGET_FPU           = ""
DISTRO_NAME          = "QorIQ SDK (FSL Reference Distro)"
ROS_DISTRO           = "galactic"
ROS_VERSION          = "2"
ROS_PYTHON_VERSION   = "3"

[log.do_compile.1383.txt](https://github.com/ros/meta-ros/files/9952235/log.do_compile.1383.txt)

razr, 2022-11-18:
it tries to generate `container_node_sizes_impl.hpp` by running `qemu-ppc -r 3.2.0 -cpu e500v2 nodesize_dbg --code --alignof "FOONATHAN_ALIGNOF(T)" container_node_sizes_impl.hpp`
Could you attach `nodesize_dbg` as well?

razr, 2022-11-18:
I have cross-compiled `foonathan-memory` on Ubuntu, and indeed `qemu-ppc` coredumps if `-cpu e500v2` is defined.

```bash
root@5ab0e7ed308e:/work/ppc-build# file ./tool/nodesize_dbg
./tool/nodesize_dbg: ELF 32-bit MSB executable, PowerPC or cisco 4500, version 1 (GNU/Linux), statically linked, BuildID[sha1]=56fae167424bc6a7951498cb5bc770e63ff9158f, for GNU/Linux 3.2.0, not stripped
```

works
```bash
qemu-ppc-static -r 3.2.0  ./tool/nodesize_dbg --code --alignof "FOONATHAN_ALIGNOF(T)" > container_node_sizes_impl.hpp
```

doesn't work
```bash
qemu-ppc-static -r 3.2.0 -cpu e500v2 ./tool/nodesize_dbg --code --alignof "FOONATHAN_ALIGNOF(T)"
qemu: uncaught target signal 4 (Illegal instruction) - core dumped
Illegal instruction (core dumped)
```

robwoolley, 2024-11-21:
This issue appears to be a problem with qemu which is out of scope of meta-ros.   Given that ROS2 Galactic, Yocto Dunfell, and the QorIQ SDK machine type aren't supported, I can't test if this is still a problem.  Please reopen if this is still a problem that we can help you with.

closed 2024-11-21 by robwoolley